MarlinFirmware/Marlin #25063: [BUG] (compile errors)

iZz0s, 2022-12-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling .pio\build\STM32F103RE_btt\src\src\feature\e_parser.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\feature\host_actions.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\feature\pause.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\feature\tmc_util.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\bedlevel\G42.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\bedlevel\M420.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\bedlevel\abl\G29.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\bedlevel\abl\M421.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\calibrate\G28.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\calibrate\M48.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\config\M200-M205.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\config\M220.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\config\M221.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\config\M301.cpp.o
Marlin\src\gcode\calibrate\M48.cpp:285: error: unterminated argument list invoking macro "TERN_"
  285 | #endif // Z_MIN_PROBE_REPEATABILITY_TEST
      |
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\config\M302.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\config\M304.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\config\M309.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\config\M92.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\control\M108_M112_M410.cpp.o
Marlin\src\gcode\calibrate\M48.cpp: In static member function 'static void GcodeSuite::M48()':
Marlin\src\gcode\calibrate\M48.cpp:115:3: error: 'TERN_' was not declared in this scope
  115 |   TERN_(HAS_PTC, ptc.set_enabled(parser.boolval('C', true));
      |   ^~~~~
Marlin\src\gcode\calibrate\M48.cpp:78:22: warning: unused variable 'raise_after' [-Wunused-variable]
   78 |   const ProbePtRaise raise_after = parser.boolval('E') ? PROBE_PT_STOW : PROBE_PT_RAISE;
      |                      ^~~~~~~~~~~
Marlin\src\gcode\calibrate\M48.cpp:111:16: warning: unused variable 'was_enabled' [-Wunused-variable]
  111 |     const bool was_enabled = planner.leveling_active;
      |                ^~~~~~~~~~~
Marlin\src\gcode\calibrate\M48.cpp:115:3: error: expected '}' at end of input
  115 |   TERN_(HAS_PTC, ptc.set_enabled(parser.boolval('C', true));
      |   ^~~~~
Marlin\src\gcode\calibrate\M48.cpp:59:24: note: to match this '{'
   59 | void GcodeSuite::M48() {
      |                        ^
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\control\M111.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\control\M120_M121.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\gcode\control\M17_M18_M84.cpp.o
*** [.pio\build\STM32F103RE_btt\src\src\gcode\calibrate\M48.cpp.o] Error 1
=============================================================================================== [FAILED] Took 11.09 seconds ===============================================================================================

Environment      Status    Duration
---------------  --------  ------------
STM32F103RE_btt  FAILED    00:00:11.093
========================================================================================== 1 failed, 0 succeeded in 00:00:11.093 ========================================================================================== 

 *  Der Terminalprozess "C:\Users\Stephan\.platformio\penv\Scripts\platformio.exe 'run'" wurde mit folgendem Exitcode beendet: 1. 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste. 

### Bug Timeline

new bug, after last pull

### Expected behavior

expected to compile like before.

### Actual behavior

compile failed.

### Steps to Reproduce

press compile.

### Version of Marlin Firmware

bugfix 2.1.1 newest

### Printer model

SV01 custom

### Electronics

SKR Mini E3 RRF

### Add-ons

bltouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-12-04:
This will be fixed in https://github.com/MarlinFirmware/Marlin/pull/25055

closed 2022-12-04 by thisiskeithb

thisiskeithb, 2022-12-04:
This is now fixed in `bugfix-2.1.x`, so pull down a fresh copy.

iZz0s, 2022-12-05:
thanks, working again :)