MarlinFirmware/Marlin #25064: [BUG] compiler warnings in STM32F103RE_btt

iZz0s, 2022-12-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Compiling .pio\build\STM32F103RE_btt\src\src\lcd\menu\menu_configuration.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\lcd\menu\menu_filament.cpp.o
In file included from C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/WString.h:29,
                 from C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/Print.h:27,
                 from C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/Stream.h:26,
                 from C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareSerial.h:29,
Compiling .pio\build\STM32F103RE_btt\src\src\lcd\menu\menu_game.cpp.o
                 from C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/WSerial.h:5,
                 from C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:47,
                 from C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:36,
                 from Marlin\src\lcd\dogm\../../inc/../HAL/./STM32/../shared/Marduino.h:36,
Compiling .pio\build\STM32F103RE_btt\src\src\lcd\menu\menu_main.cpp.o
                 from Marlin\src\lcd\dogm\../../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin\src\lcd\dogm\../../inc/../HAL/HAL.h:30,
                 from Marlin\src\lcd\dogm\../../inc/MarlinConfig.h:31,
                 from Marlin\src\lcd\dogm\dogm_Statusscreen.h:31,
                 from Marlin\src\lcd\dogm\status_screen_DOGM.cpp:32:
Marlin\src\lcd\dogm\../../module/../libs/duration_t.h: In static member function 'static void MarlinUI::draw_status_screen()':
Compiling .pio\build\STM32F103RE_btt\src\src\lcd\menu\menu_media.cpp.o
Marlin\src\lcd\dogm\../../module/../libs/duration_t.h:161:30: warning: '%02hu' directive writing 2 bytes into a region of size between 0 and 4 [-Wformat-overflow=]
Compiling .pio\build\STM32F103RE_btt\src\src\lcd\menu\menu_mixer.cpp.o
  161 |       sprintf_P(buffer, PSTR("%hud %02hu:%02hu"), d, h % 24, m);  // 1d 23:45
C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/avr/pgmspace.h:71:40: note: in definition of macro 'sprintf_P'
   71 | #define sprintf_P(s, ...) sprintf((s), __VA_ARGS__)
      |                                        ^~~~~~~~~~~
Marlin\src\lcd\dogm\../../module/../libs/duration_t.h:161:25: note: in expansion of macro 'PSTR'
  161 |       sprintf_P(buffer, PSTR("%hud %02hu:%02hu"), d, h % 24, m);  // 1d 23:45
      |                         ^~~~
Marlin\src\lcd\dogm\../../module/../libs/duration_t.h:161:30: note: directive argument in the range [0, 59]
  161 |       sprintf_P(buffer, PSTR("%hud %02hu:%02hu"), d, h % 24, m);  // 1d 23:45
C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/avr/pgmspace.h:71:40: note: in definition of macro 'sprintf_P'
   71 | #define sprintf_P(s, ...) sprintf((s), __VA_ARGS__)
Compiling .pio\build\STM32F103RE_btt\src\src\lcd\menu\menu_motion.cpp.o
      |                                        ^~~~~~~~~~~
Marlin\src\lcd\dogm\../../module/../libs/duration_t.h:161:25: note: in expansion of macro 'PSTR'
  161 |       sprintf_P(buffer, PSTR("%hud %02hu:%02hu"), d, h % 24, m);  // 1d 23:45
      |                         ^~~~
C:\Users\Stephan\.platformio\packages\framework-arduinoststm32\cores\arduino/avr/pgmspace.h:71:34: note: 'sprintf' output between 9 and 13 bytes into a destination of size 10
   71 | #define sprintf_P(s, ...) sprintf((s), __VA_ARGS__)
      |                           ~~~~~~~^~~~~~~~~~~~~~~~~~
Marlin\src\lcd\dogm\../../module/../libs/duration_t.h:161:7: note: in expansion of macro 'sprintf_P'
  161 |       sprintf_P(buffer, PSTR("%hud %02hu:%02hu"), d, h % 24, m);  // 1d 23:45
      |       ^~~~~~~~~
Compiling .pio\build\STM32F103RE_btt\src\src\lcd\menu\menu_probe_offset.cpp.o
Compiling .pio\build\STM32F103RE_btt\src\src\lcd\menu\menu_temperature.cpp.o

### Bug Timeline

since a week around

### Expected behavior

compiling without this warnings, anyways, the compile running through and SUCCESS.

### Actual behavior

got this mentioned errors on compile. 

### Version of Marlin Firmware

bugfix 2.1.1 newest

### Printer model

SV01 custom

### Electronics

SKR Mini E3 RRF

### Add-ons

bltouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[image]

borland1, 2022-12-05:
Sorry your having trouble.   This is where you report issues with the Marlin repository.  There is another place to report issues with PlatformIO.   Sounds like you just need some help.  Try asking here for help...

[PlatformIO Community](https://community.platformio.org/)

You can report a PlatformIO bug/issues here..

[PlatformIO Issues](https://github.com/platformio/platformio-core/issues)

thisiskeithb, 2022-12-05:
This will be fixed in https://github.com/MarlinFirmware/Marlin/pull/24879

closed 2022-12-05 by thisiskeithb