MarlinFirmware/Marlin #25067: BUG Probing Margin must be greater or equal to probe offset

jgbrown54, 2022-12-06:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Probing Margin cannot be less than the positive probe offset. This causes the carriage to reach the maximum or minimum X or Y causing a nonuniform probing matrix. For example, if the probe offset is -40 and the probing margin is 10, and the bed size is 300X300, the probe point on the left of the bed is at 10 but the probe point on the right of the bed is 300 - 40 or 260 because the nozzle is not allowed to go beyond the bed size of 300. If the probe offset is +40, the same would happen on the left of the bed. This would require a modification of the Probe.h module to account for this. It should account for the highest X or Y probe offset and force the Min_X, Max_X, Min_Y, & Max_Y to account for the probe offset so that the margin is equal on all sides.

### Bug Timeline

Existing issue.

### Expected behavior

I expect bed to be probed logical equal-distance probe points so that extrapolation is more acurate.

### Actual behavior

On my printer, the probe offset is at X -40 and the bed size is 300X300. When the bed is probed with a 5X5 matrix, the spacing of the probed X coordinates is 70mm or 10, 80, 150, 220, 290 however the nozzle reaches the right edge (300mm) and the probe is at 250mm. This occurs for the entire right column of probe points. With an equal margin on each side based on the probe offset, the probe points should be 44mm apart at 40, 84, 128, 172, 216 resulting in a square matrix on the bed allowing for more accurate interpolation of the bed leveling.

### Steps to Reproduce

1. Set probe margin at 10.
2. Set probe offset at -40.
3. Run automatic bed leveling.
4. Notice that probed points on the bed are not equal distant apart as the right column is squeezed in due to the maximum bed size.

### Version of Marlin Firmware

2.1.1

### Printer model

Longer3d LK5 Pro

### Electronics

Stock

### Add-ons

Dual blower forces probe offset to be -40 but problem exists any time probe offset is greater in either + or - than probe margin.

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

Repetier Host

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

There are configuration changes made to account for the Longer LK5 Pro. The only applicable changes are a bed size of 300X300, a probe offset of -40 X, and a probe margin of 10.

ellensp, 2022-12-06:
Margin is applied around entire edge of bed
So it doesn't work as you say it does

You list X offset -40, margin is 10, bed 300X300

But where is X_MIN_POS, Y_MIN_POS, X_MAX_POS and Y_MAX_POS ?  these are the values tar are actually used

Presuming 0,0,300,300 for the missing values
Then the X probable distance is 10 to 250 (300-10-40)  
5 points would be at (250-10)/4  = 60mm apart

so the X probe points are at 10, 70, 130,, 190 and 250

jgbrown54, 2022-12-07:
I did a test and measured the spacing. And of course, you are correct. I had a user of my config complain that not all of the bed was probed. I noticed that when probing to the right of the bed, the probing was stopped short and I thought it was because of the probe offset. And the other spacing looked even. But I didn't measure it. I do see that the matrix is evenly spaced but it is offset on the bed. Will this cause an issue with interpolation of the level on the right hand edge or should the matrix be centered? If not, you can close this issue. It took longer to write up what I thought was happening than it did to measure the spacing to verify it. Sorry.

closed 2022-12-07 by jgbrown54