MarlinFirmware/Marlin #25068: [BUG] Y Axis moves past limit on "Change Filament" command

GJSchaller, 2022-12-06:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

Ender 3 Pro, BTT SKR mini E3 v2.0, Marlin Bugfix-2.1.x a few months old (but after 2.1.1).  When initiating a Change Filament command mid-print, the printer successfully pauses, moves X to 0, but then advances Y past the limit of the bed, causing the position of the print / Y to change when the print resumes.

### Bug Timeline

New, noticed it for the first time last night.

### Expected behavior

Either for Y to limit itself to the maximum range of movement, or for Y to home itself before moving the bed forward.

### Actual behavior

The bed moved forward past the limit, hitting the end of the rails and grinding the motor, changing the position of Y when the print resumed.

### Steps to Reproduce

1. Print something with a large Y size, or at the back of the plate.
2. Initiate Change Filament from the console mid-print
3. Watch to see if the bed tries to advance past the maximum value.

### Version of Marlin Firmware

2.1.1-Bugfix (Dated Nov. 09, 2022)

### Printer model

Ender 3 Pro

### Electronics

BTT SKR mini E3 v2.0

### Add-ons

BLTouch

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration-GJSchaller.zip](https://github.com/MarlinFirmware/Marlin/files/10167589/Configuration-GJSchaller.zip)

GJSchaller, 2022-12-06:
@thisiskeithb - looking back at this, I'd forgotten I overwrote my configs on a recent pull.  Advanced_Pause_Feature is enabled in the BTT SKR E3 mini V2 by default.  My only thought / comment here is to change:

#define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MAX_POS - 10), 20 }

to be

#define NOZZLE_PARK_POINT { (X_MIN_POS + 10), (Y_MIN_POS + 10), 20 }

It's minor, but it will prevent the bed from trying to move forward past its physical limit.

thisiskeithb, 2022-12-06:
If your bed runs into `Y_MAX_POS`, that would mean `Y_MAX_POS` is not defined correctly in your config / your travel limits in general are not correct.

[Give this guide by Manuel McLure](https://manuelmclure.github.io/ConfiguringLeveling.html#determining-machine-limits-on-a-printer-that-homes-to-xminymin) a read to properly setup your travel limits.