MarlinFirmware/Marlin #25069: [BUG] Extruder stops extruding, Linear advanced and TMC 2208

victoroos, 2022-12-06:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

hello,

So, after I read that the TMC 2208 can finally be used in the extruder axis with linear advanced. I got it all setup and calibrated nicely.
Now trying a few prints, but my last 2 prints stopped extruding at a certain point. THe extruder stepper motor just stops rotating.

I tried different files, checking the gcode, the extruder definitely gets commands in the GCODe: https://pastebin.com/YmQEneJp

### Expected behavior

I expected it to finish printing

### Actual behavior

It stopped rotating

### Version of Marlin Firmware

2.1.1

### Printer model

TEvo Tarantula PRo

### Electronics

MKS L 

### Add-ons

TMC 2208

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

hello,

So, after I read that the TMC 2208 can finally be used in the extruder axis with linear advanced. I got it all setup and calibrated nicely.
Now trying a few prints, but my last 2 prints stopped extruding at a certain point. THe extruder stepper motor just stops rotating.

I tried different files, checking the gcode, the extruder definitely gets commands in the GCODe: https://pastebin.com/YmQEneJp

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10170203/Configuration.zip)

victoroos, 2022-12-06:
It looks like printing a calibration cube works. It is very file specific.

victoroos, 2022-12-07:
Tested with the bug fix, still same problem with the gcode "attached"

victoroos, 2022-12-07:
LA is now off and it works :/

victoroos, 2022-12-07:
LA is off and it works :)

victoroos, 2022-12-07:
LA is off and it works :/

jhstatewide, 2022-12-07:
I have the same bug, TMC2208 running in standalone. I read the release notes that TMC2208 was supported for LINEAR_ADVANCE now with 2.1 so attempted an upgrade. The print will start just fine but 5-10 minutes in, extruder halts entirely. I then flipped linear advance off and get the same behavior. Linear advance on/off, it does not matter. I'm using a FLSUN QQS-PRO, opened an issue in one of the forks:

https://github.com/Foxies-CSTL/Marlin_2.1.x/issues/12

Printer works fine on TMC2208 on Marlin 2.0.8, no linear advance.

No combination of settings for Marlin 2.1 appear to work properly, so holding on Marlin 2.0.8 for now.

victoroos, 2022-12-07:
I'm running the TMC in UART mode. Weird that you have it as well.

Nuck-TH, 2022-12-08:
I too have random overcurrent errors(regardless of set current, be it 500mArms(stock) or 1Arms(nominal for motors), but on XY axis instead of extruder.
In the same vein as above, i haven't had this issue on 2.0.x branch.
Kingroon KP3S v1.3 board with soldered-on TMC2225 in uart mode.

LaurenceGough, 2022-12-24:
Same issue here with a Creality 4.2.2 board running Marlin 2.1.2 (built today).

K factor gcode test print works fine but when printing a real print it failed a few layers in.

Extruder is able to be moved by hand, all other steppers are still active.

LaurenceGough, 2022-12-24:
> I have the same bug, TMC2208 running in standalone. I read the release notes that TMC2208 was supported for LINEAR_ADVANCE now with 2.1 so attempted an upgrade. The print will start just fine but 5-10 minutes in, extruder halts entirely. I then flipped linear advance off and get the same behavior. Linear advance on/off, it does not matter. I'm using a FLSUN QQS-PRO, opened an issue in one of the forks:
> 
> [Foxies-CSTL#12](https://github.com/Foxies-CSTL/Marlin_2.1.x/issues/12)
> 
> Printer works fine on TMC2208 on Marlin 2.0.8, no linear advance.
> 
> No combination of settings for Marlin 2.1 appear to work properly, so holding on Marlin 2.0.8 for now.

I am experiencing the exact same issues. LA on or off. A couple of layers in and my Extruder motor goes completely dead. Only turning the printer off and back on resolves the issue. No extruder stepper moves etc will get it going.

With Marlin 2.0.8.26 (Creality fork) I never had an issue. 

However I'm sure there are 1000's of people using Marlin 2.1.x without issue, I don't get it.

Jumbo125, 2023-02-12:
Does AL work with TMC 2208 driver when I use marlin 2.1.1?
So I will be compile the new Bugfix version.

Or exists the bug?

Postman05, 2023-03-03:
I had the same problem with my Tevo Tarantula Pro + TMC2208 (standalone) on Marlin 2.0.8. The problem disappeared when I disabled S-curve.
Maybe this will help you too.

thisiskeithb, 2023-04-26:
Please download [`bugfix-2.1.x`](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.1.x.zip) to test with the latest code and let us know if you're still having this issue.

jhstatewide, 2023-04-26:
@thisiskeithb Awesome... can you provide some detail? Was there a certain commit or feature that landed that makes this issue likely fixed?

thisiskeithb, 2023-04-26:
> Was there a certain commit or feature that landed that makes this issue likely fixed?

[Several Linear Advance PRs have been merged](https://github.com/MarlinFirmware/Marlin/pulls?q=is%3Apr+label%3A%22F%3A+Linear+Advance%22+merged%3A2022-12-06..2023-04-26+) since this issue was opened, which is why I asked users experiencing this bug to download the latest `bugfix-2.1.x` and report back.

jhstatewide, 2023-04-26:
@thisiskeithb Thanks for the detail!

thisiskeithb, 2023-06-26:
Fixed in the various PRs mentioned above.

closed 2023-06-26 by thisiskeithb