MarlinFirmware/Marlin #25072: Homing feedrate and homing bump divisor ignored

GameFan01, 2022-12-07:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

The title says it all: I have my max feedrates set to 150mm/s for X and Y, and 25mm/s for Z.
I set the Z homing feedrate to 20mm/min and the Z homing bump divisor to something like 32. 
The outcome was that the printer homed all axes at the maximum feedrates and completely ignored the homing feedrates and bump divisors. Only Homing_bump_mm setting works as intended (tested by setting all to 100mm).

I've then tried to set the bump divisor to 255 (the highest value allowed in that variable) - no effect
I've tried setting all the homing feedrates to 1mm/min - no effect
Finally I set the max feedrates to around 1/10th the previous value - the printer homed at those feedrates as it should.

### Bug Timeline

The bug is a new issue, I've noticed it just now (7.12.2022 at 19:36

### Expected behavior

The printer should home at the feedrates set in HOMING_FEEDRATE_MM_M and slow down according to BUMP_DIVISOR

### Actual behavior

The printer homes at MAX_FEEDRATE

### Steps to Reproduce

I have no clue. 
I'm updating the firmware over Octoprint, so it might have something to do with it, although I would be surprised.
I'm not setting any feedrates/acceleration on startup, so that can't be it, and it behaves the same immediately after the update, so the slicer is definitely out of the equation.

### Version of Marlin Firmware

Marlin 2.1.1 Stable release

### Printer model

Ender 3

### Electronics

MKS gen L v1.0, TMC2208 and a reprap full graphic smart controller 

### Add-ons

Raspberry pi 3B+

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10179877/Configuration.zip)

imaluschi, 2023-02-04:
No, it hasn't something to do with Octoprint.
Exactly the same Problem here, Marlin-2.1.2 (Dec 18, 2022), built with Visual Studio Code.
Hardware: MKS Sgen, TMC2209 (X and Y, UART Mode), DVR8825 (Z1 and Z2) and reprap full graphic smart controller
No matter which values are given, homing always with the same feedrate, same behaviour for all axes (X, Y and Z).

imaluschi, 2023-02-04:
Seems to be solved in the actual code - bugfix-2.1.x.

GameFan01, 2023-02-04:
Yea, I upgraded recently and the bug is gone. I completely forgot about this issue. Sorry about that.

closed 2023-02-04 by GameFan01