MarlinFirmware/Marlin #25075: [BUG] Marlin will not compile with 2 extruders while utilizing MPCTemp

DWillems26, 2022-12-08:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

While attempting to verify/compile Marlin in Arduino IDE I receive a failure error due to static assertions with MPCTemp enabled and 2 extruders. All 6 static assertions in settings.cpp fail saying I must have HOTENDS items. While compiling with PID Temp enabled instead of MPC Temp I receive no errors (with extruders set to 2). When I change extruders to 1 with MPCTemp enabled I receive no errors and Marlin does finish compiling.

I receive the same error with Marlin 2.0.9.5.

It may be that I simply missed a setting?

[Marlin 2.1.x config files for reference.zip] (https://github.com/MarlinFirmware/Marlin/files/10188767/Marlin.2.1.x.config.files.for.reference.zip)

See Arduino IDE error message below...

C:\...\Marlin-2.1.x\Marlin\src\module\settings.cpp: In static member function 'static void MarlinSettings::reset()':
C:\...\Marlin-2.1.x\Marlin\src\module\settings.cpp:3300:5: error: static assertion failed: MPC_HEATER_POWER must have HOTENDS items.
     static_assert(COUNT(_mpc_heater_power) == HOTENDS, "MPC_HEATER_POWER must have HOTENDS items.");
     ^~~~~~~~~~~~~

C:\...\Marlin-2.1.x\Marlin\src\module\settings.cpp:3301:5: error: static assertion failed: MPC_BLOCK_HEAT_CAPACITY must have HOTENDS items.
     static_assert(COUNT(_mpc_block_heat_capacity) == HOTENDS, "MPC_BLOCK_HEAT_CAPACITY must have HOTENDS items.");
     ^~~~~~~~~~~~~

C:\...\Marlin-2.1.x\Marlin\src\module\settings.cpp:3302:5: error: static assertion failed: MPC_SENSOR_RESPONSIVENESS must have HOTENDS items 

     static_assert(COUNT(_mpc_sensor_responsiveness) == HOTENDS, "MPC_SENSOR_RESPONSIVENESS must have HOTENDS items.");
     ^~~~~~~~~~~~~

C:\...\Marlin-2.1.x\Marlin\src\module\settings.cpp:3303:5: error: static assertion failed: MPC_AMBIENT_XFER_COEFF must have HOTENDS items.
     static_assert(COUNT(_mpc_ambient_xfer_coeff) == HOTENDS, "MPC_AMBIENT_XFER_COEFF must have HOTENDS items.");
     ^~~~~~~~~~~~~

C:\...\Marlin-2.1.x\Marlin\src\module\settings.cpp:3305:7: error: static assertion failed: MPC_AMBIENT_XFER_COEFF_FAN255 must have HOTENDS items.
       static_assert(COUNT(_mpc_ambient_xfer_coeff_fan255) == HOTENDS, "MPC_AMBIENT_XFER_COEFF_FAN255 must have HOTENDS items.");
       ^~~~~~~~~~~~~

C:\...\Marlin-2.1.x\Marlin\src\module\settings.cpp:3307:5: error: static assertion failed: FILAMENT_HEAT_CAPACITY_PERMM must have HOTENDS items.
     static_assert(COUNT(_filament_heat_capacity_permm) == HOTENDS, "FILAMENT_HEAT_CAPACITY_PERMM must have HOTENDS items.");
     ^~~~~~~~~~~~~

Using library U8glib-HAL at version 0.5.2 in folder: C:\...\Arduino\libraries\U8glib-HAL 
exit status 1

Compilation error: static assertion failed: MPC_HEATER_POWER must have HOTENDS items.

### Bug Timeline

New bug. This is my first time utilizing Marlin 2.0.9.5 and above.

### Expected behavior

I expected Marlin to compile without errors.

### Actual behavior

Static Assertion failures.

### Steps to Reproduce

Referencing Marlin 2.1 Bug Fix. In configuration.h enable MPC Temp (line 652), disable PIDTemp, enable MPC_FAN_0_ALL_HOTENDS (line 703) and set extruders to "2" (line 230).

### Version of Marlin Firmware

2.1.x

### Printer model

Modix Big 120x

### Electronics

stock electronics. MKS Gen V1.4 8-bit board.

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

No other modifications were done outside of configuration.h and configuration.adv.h.

thisiskeithb, 2022-12-08:
Please try [compiling in VSCode with PlatformIO](https://marlinfw.org/docs/basics/install_platformio_vscode.html) and report back.

DWillems26, 2022-12-09:
> Please try [compiling in VSCode with PlatformIO](https://marlinfw.org/docs/basics/install_platformio_vscode.html) and report back.

Thank you for that link. I couldn't figure out how to get VSCode to build Marlin before. Unfortunately I had the same results. Same static assertion fails with 2 extruders, but the build succeeded with only 1 extruder. Using Marlin 2.1.x and the bugfix version as well.

But... SUCCESS!! Sort of... If I replace HOTENDS with "1" in settings.cpp the build is successful even though I have extruders set to 2. This is of course done for all 6 static assertions. I know this is not the proper method though. And I don't know enough about writing code to know if this will damage anything. I definitely don't know enough to write the proper code. Will simply replacing HOTENDS with 1 suffice for now as a temporary solution? Or do I risk damaging anything?

For example... 
Original code: "static_assert(COUNT(_mpc_heater_power) == HOTENDS, "MPC_HEATER_POWER must have HOTENDS items."
Modified code: "static_assert(COUNT(_mpc_heater_power) == 1, "MPC_HEATER_POWER must have HOTENDS items."

Thank You for the assistance!

....

Marlin\src\module\settings.cpp: In static member function 'static void MarlinSettings::reset()':
Marlin\src\module\settings.cpp:3300:5: error: static assertion failed: MPC_HEATER_POWER must have HOTENDS items.
     static_assert(COUNT(_mpc_heater_power) == HOTENDS, "MPC_HEATER_POWER must have HOTENDS items.");
     ^~~~~~~~~~~~~
Marlin\src\module\settings.cpp:3301:5: error: static assertion failed: MPC_BLOCK_HEAT_CAPACITY must have HOTENDS items.
     static_assert(COUNT(_mpc_block_heat_capacity) == HOTENDS, "MPC_BLOCK_HEAT_CAPACITY must have HOTENDS items.");
     ^~~~~~~~~~~~~
Marlin\src\module\settings.cpp:3302:5: error: static assertion failed: MPC_SENSOR_RESPONSIVENESS must have HOTENDS items.
     static_assert(COUNT(_mpc_sensor_responsiveness) == HOTENDS, "MPC_SENSOR_RESPONSIVENESS must have HOTENDS items.");
     ^~~~~~~~~~~~~
Marlin\src\module\settings.cpp:3303:5: error: static assertion failed: MPC_AMBIENT_XFER_COEFF must have HOTENDS items.
     static_assert(COUNT(_mpc_ambient_xfer_coeff) == HOTENDS, "MPC_AMBIENT_XFER_COEFF must have HOTENDS items.");
     ^~~~~~~~~~~~~
Marlin\src\module\settings.cpp:3305:7: error: static assertion failed: MPC_AMBIENT_XFER_COEFF_FAN255 must have HOTENDS items.
       static_assert(COUNT(_mpc_ambient_xfer_coeff_fan255) == HOTENDS, "MPC_AMBIENT_XFER_COEFF_FAN255 must have HOTENDS items.");
       ^~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\mega2560\src\src\module\settings.cpp.o] Error 1
=========================================================== [FAILED] Took 36.73 seconds ===========================================================

Environment    Status    Duration
-------------  --------  ------------
mega2560       FAILED    00:00:36.732
====================================================== 1 failed, 0 succeeded in 00:00:36.732 ====================================================== 

 *  The terminal process "C:\Users\...\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'mega2560'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

ellensp, 2022-12-09:
you have two hotends, thus you need two sets of parameters for them, like this.
```CPP

  #define MPC_HEATER_POWER { 40.0f, 40.0f }                  // (W) Heat cartridge powers.

  #define MPC_INCLUDE_FAN                             // Model the fan speed?

  // Measured physical constants from M306
  #define MPC_BLOCK_HEAT_CAPACITY { 16.7f, 16.7f }           // (J/K) Heat block heat capacities.
  #define MPC_SENSOR_RESPONSIVENESS { 0.22f, 0.22f }         // (K/s per ∆K) Rate of change of sensor temperature from heat block.
  #define MPC_AMBIENT_XFER_COEFF { 0.068f, 0.068f }           // (W/K) Heat transfer coefficients from heat block to room air with fan off.
  #if ENABLED(MPC_INCLUDE_FAN)
    #define MPC_AMBIENT_XFER_COEFF_FAN255 { 0.097f, 0.097f }  // (W/K) Heat transfer coefficients from heat block to room air with fan on full.
  #endif

  // For one fan and multiple hotends MPC needs to know how to apply the fan cooling effect.
  #if ENABLED(MPC_INCLUDE_FAN)
    #define MPC_FAN_0_ALL_HOTENDS
    //#define MPC_FAN_0_ACTIVE_HOTEND
  #endif

  #define FILAMENT_HEAT_CAPACITY_PERMM { 5.6e-3f, 5.6e-3f }    // 0.0056 J/K/mm for 1.75mm PLA (0.0149 J/K/mm for 2.85mm PLA).

```

closed 2022-12-09 by ellensp