MarlinFirmware/Marlin #25076: [BUG] (bug summary)

bihailantian655, 2022-12-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

stm32f103cbt6

### Bug Timeline

first

### Version of Marlin Firmware

2.1.1

### Printer model

klipper

### Electronics

klipper board

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

i  flashed the  marlin firmware  on  my klipper 3d pritn  motherboard
but  computer can not reconize  marlin.there  no  info sign from the  mother board  

just  modifyed the following  config

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_MALYAN_M200
#endif
#define SERIAL_PORT 1

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32F103CB_malyan
include_dir  = Marlin
extra_configs =
    Marlin/config.ini
    ini/avr.ini

ellensp, 2022-12-09:
Description is unclear

is this a actual BOARD_MALYAN_M200 or just something with the same MPU?

bihailantian655, 2022-12-09:
not actual BOARD_MALYAN_M200     just  the  same  mpu

ellensp, 2022-12-09:
Well just because it has the same MPU does not mean its the same.. the stm32 is way to configurable for that

And not supporting some new controller is not a bug at best its a feature request.

How about starting with updating to a feature request and posting details on this controller?

closed 2022-12-09 by ellensp