MarlinFirmware/Marlin #25077: [BUG] (Custom Menus don't work)

threed-factory-store, 2022-12-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I added custom menu items to my Configuration_adv.h file, using the [examples](https://marlinfw.org/docs/configuration/configuration.html#custom-user-menu-items) as a template:

#define CUSTOM_MENU_MAIN
#if ENABLED(CUSTOM_MENU_MAIN)
  #define CUSTOM_MENU_MAIN_TITLE "Leveling"
  #define CUSTOM_MENU_MAIN_SCRIPT_DONE "M117 User Script Done"
  #define CUSTOM_MENU_MAIN_SCRIPT_AUDIBLE_FEEDBACK
  //#define CUSTOM_MENU_MAIN_SCRIPT_RETURN   // Return to status screen after a script
  #define CUSTOM_MENU_MAIN_ONLY_IDLE         // Only show custom menu when the machine is idle

  #define MAIN_MENU_ITEM_1_DESC "Level for " PREHEAT_1_LABEL
  #define MAIN_MENU_ITEM_1_GCODE "M190 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM109 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND) " \nG28 \nG29 P1 \nG29 P3 \nG29 S1 \nG29 F 10 \nG29 A \nM500 \nM140 S0 \nM104 S0"
  //#define MAIN_MENU_ITEM_1_CONFIRM          // Show a confirmation dialog before this action

  #define MAIN_MENU_ITEM_2_DESC "Level for " PREHEAT_2_LABEL
  #define MAIN_MENU_ITEM_2_GCODE "M190 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM109 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND) " \nG28 \nG29 P1 \nG29 P3 \nG29 S2 \nG29 F 10 \nG29 A \nM500 \nM140 S0 \nM104 S0"
  //#define MAIN_MENU_ITEM_2_CONFIRM

  #define MAIN_MENU_ITEM_3_DESC "Home & UBL Info"
  #define MAIN_MENU_ITEM_3_GCODE "G28\nG29 W"
#endif

Nothing shows up on my printer's screen.  

I know my firmware is being loaded because I add my build number to my CUSTOM_MACHINE_NAME, and the correct build number is being displayed when the printer boots.

### Bug Timeline

new

### Expected behavior

I expect my "Leveling" menu item to show up on my printer's main screen, with three items displayed when I select the Leveling item.

### Actual behavior

No "Leveling" menu is displayed.

### Steps to Reproduce

1. Modify Configuration_adv.h as shown above.
2. Modify CUSTOM_MACHINE_NAME in Configuration.h
3. Build
4. Boot printer from new firmware.

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Printer model

Creality Ender 3 V2

### Electronics

Stock

### Add-ons

Inductive bed sensor

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10195720/Configuration.zip)

thisiskeithb, 2022-12-09:
I don't think Custom Menus work in anything other than MarlinUI-based UIs. In your case, that'd be `DWIN_MARLINUI_PORTRAIT` or `DWIN_MARLINUI_LANDSCAPE`.

thisiskeithb, 2022-12-09:
I've added a sanity check for now: https://github.com/MarlinFirmware/Marlin/pull/25079

Can you rename this to a Feature Request?

thinkyhead, 2022-12-18:
This kind of exists as a standing TODO for all third-party custom screens, so we don't necessarily need an open FR for it.

closed 2022-12-18 by thisiskeithb