MarlinFirmware/Marlin #25078: [BUG] Wanhao Dub 4S Nozzle E1 not heating up, after flashing to Marlin

CoolZeroNL, 2022-12-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

E1 Nozzle is not heating up, after flashing to latest (c16d3655bf) commit of bugfix-2.1.x branch. E2 however is working.

With the Sailfish 7.7 firmware  E1 and E2 are heating up.

When i flash the machine to Marlin then E1 wont heat up, and fail after timeout.

I tried also the firmware with no change to the configuration files, but also this does not heat E1. 

Stock configuration used (copy fom: https://github.com/MarlinFirmware/Configurations/tree/import-2.1.x/config/examples/Wanhao/Duplicator%204S) :
    - https://github.com/CoolZeroNL/Marlin-Build/blob/main/configuration/wanhao_dublicator_4s/bugfix-2.1.x/Configuration.h
    - https://github.com/CoolZeroNL/Marlin-Build/blob/main/configuration/wanhao_dublicator_4s/bugfix-2.1.x/Configuration_adv.h

Own adjusted configuration:
    - https://github.com/CoolZeroNL/Marlin-Build/blob/main/configuration/wanhao_dublicator_4s/bugfix-2.1.x/Configuration.h_own
    - https://github.com/CoolZeroNL/Marlin-Build/blob/main/configuration/wanhao_dublicator_4s/bugfix-2.1.x/Configuration_adv.h_own

Firmware build pipeline (dont know i u can access but):
    - Compile log: https://github.com/CoolZeroNL/Marlin-Build/actions/runs/3657923466/jobs/6182065241

ps. Electric wiring is checked, and all is Oke. Also when i flash the machine back to Sailfish 7.7 E1 and E2 are heating up.

### Bug Timeline

First version of marlin on this machine

### Expected behavior

Nozzle should heat up to tempature

### Actual behavior

Nothing happens, after timeout message: Heating failed: E1 printer halted, please reset

### Steps to Reproduce

Steps:
1. Menu -> Tempature -> Nozzle 1 --> 50 degree

### Version of Marlin Firmware

Commit: c16d3655bf from bugfix-2.1.x

### Printer model

Wanhao Dublicator 4S

### Electronics

stock

### Add-ons

none

### Bed Leveling

No Bed Leveling

### Your Slicer

Simplify3D

### Host Software

Same as my slicer

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-12-09:
The BOARD_MIGHTYBOARD_REVE is special, it uses pin that are not in the arduino build env.
Use build env MightyBoard1280

CoolZeroNL, 2022-12-10:
Thats it, great thanks !

closed 2022-12-10 by CoolZeroNL