MarlinFirmware/Marlin #25082: [BUG] (changes made in configuration->advance settings ->temperature are not persisted after save and cold boot)

tutdeveloper, 2022-12-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

In the LCD menus, when navigating to Configuration, then Advanced Settings, then Temperature, you can set the Min and Max temp as well as Autotemp and Fact. If I change Min or Max or set Autotemp to On, and then perform a Store Settings (with beep to confirm), the temp settings does not seem to be saved to the EEPROM. Although they will remain for the current session, if I perform a cold boot, the settings will revert back to their previous values.

I've tried this on two different machines and have also compiled and deployed a bugfix-2.1.x to confirm.
Max temp in my Configuration.h is set to 290, however, the menu in navigation always defaults back to 250. I can the set it to a max of 275 in the menu and save the settings, however, after a cold boot, the setting is back to 250 (same pattern for min temp and autotemp)

I've also performed an Initialize EEPROM as well as a Restore Defaults to make sure that there is no issues with the EEPROM and I am able to change and store other settings.

### Expected behavior

When performing a Store Settings in the configuration menu, I expect the Temperature settings to be stored and persisted after a cold boot.

### Actual behavior

Temperature settings under Configuration->Advance Configuration->Temperature always resets back to a default set of values after a cold boot in stead of new values that the user have selected and saved

### Steps to Reproduce

In LCD menus,
1) Navigate to Configuration
2) Navigate to Advanced Settings
3) Navigate to Temperature
4) change a value for Min Temp, Max temp or Autotemp
5) Navigate to Configuration
6) Select Store Settings
7) Switch printer off
8) Switch printer on
9) Navigate back to the Temperature screen in advanced Settings
10) Observe that the temperature settings previously made was not retained.

### Version of Marlin Firmware

2.1.1 as well as bugfix-2.1.x

### Printer model

Custom

### Electronics

BTT SKR Mini E3 v2.0

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10201805/Configuration.zip)
The attached config files are for marlin-bugfix-2.1.x, however, marlin-2.1.1 exhibits the exact same behavior.

ellensp, 2022-12-13:
Correct these are not saved.  This has never been savable. This by design.

ellensp, 2022-12-13:
Please give this a try https://github.com/MarlinFirmware/Marlin/pull/25093

thisiskeithb, 2022-12-16:
Please post your testing/findings in https://github.com/MarlinFirmware/Marlin/pull/25093

Closing since a PR has been created.

closed 2022-12-16 by thisiskeithb