MarlinFirmware/Marlin #25084: Settings to BLTouch

Puch-tdi, 2022-12-11:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The standard installation of BLTouch fails on some boards because the hardware works the other way around.
I found and resolved the problem and will place a thread on github later on but maybe you can change this issue in your setting (if you think its unsafe place a warning please).
I have a home build Delta with a Velleman Vertex K8400 board, enlarged bed, two titan extruders and now a genuine BLTouch.
I have it finally running on Marlin 2.1.1 (arduino and also VSC having a lot of trouble to build on this board).
First my normal config and later on i added the probe.
But 3DRAG (Velleman is a clone) boards work the other way around on the endstops (specially the digital endstops factory-fitted from Velleman).
You need to set Z_MIN_PROBE_ENDSTOP_INVERTING to "true" but then it will not allow you to upload because of an error telling you to set it to false.
Same goes for Z_MIN_ENDSTOP_INVERTING.
It is telling that BLTouch require it to set it to false but then the Z_MIN can't read the connection and it fails to home or probe.
My solution is this:
In the Conditionals_LCD.h you can change this by commend it out // in front of the error rule (see attached file), then it will allow to upload and now my printer is fully functional including all options like safe homing etc.
There are loads of people having a headache because of this problem, is it possible to allow this change in the next release so it does allow the setting we need?
[image]
[image]

Please don't mind the "Motherboard not defined"error, somehow it tell me this every time but it does allow me to upload, the same reason it took me a while to build my config (VSC can not auto-build on this mainbord).

### Bug Timeline

new

### Expected behavior

i expected it to work out of the box and the marlin help pages to configure it.
For now it works for me.

### Actual behavior

it failed to home and probe, didn't see the endstop

### Steps to Reproduce

build V2.1.1 on 3drag or Velleman board and try adding the BLTouch

### Version of Marlin Firmware

2.1.1

### Printer model

K8400 Vertex Custom

### Electronics

Stock mainboard, two Titan extruders (one mirrored), stock LED print, Dual Z-axis upgrade (one motor and Belt-drive), stock LCD with SD reader

### Add-ons

BLTouch V3.1 (pin 53 for servo0), Other belts on GT2 and upgraded Z-Drive

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Slic3r

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/10202482/Config.zip)
Build with the latest version of VSC

Puch-tdi, 2022-12-11:
Forgot to mention i tried everything first from BLTouch manufacurer, Putting the probe in the 5V configuration will also not function standard. Now it works i didn't trie again in the 5V mode.

EvilGremlin, 2022-12-11:
Error is right,if your board circuitry don't allow it - you must choose another pin for probe (or change circuitry)

EvilGremlin, 2022-12-11:
Adding to check for one oddball board... yeah it's easy but ewww

Puch-tdi, 2022-12-11:
> Error is right,if your board circuitry don't allow it - you must choose another pin for probe (or change circuitry)

There is the issue, the board is not equiped with other pins.

Puch-tdi, 2022-12-11:
indead, nothing free at all. i had to solder the wire for the servo out of the flatcable for the lcd

closed 2022-12-12 by thinkyhead