MarlinFirmware/Marlin #25086: Ability to pause at job end and repeat same job.

plundrigan, 2022-12-12:
### Is your feature request related to a problem? Please describe.

The ability to repeat the same print x times, with a pause in-between to allow for bed clearing. 

### Describe the feature you want

It would be nice to have the possibly of repeat job. 

You would start a print. (It could prompt for # of prints)
The print would pause instead of stop and shutdown after first completion.
You clear the bed of your # finished part.
Click un-pause and the same job starts back up again
this process would repeat for 3 prints. 

This method would allow for the creation of multiple same jobs in a row. 

I am new to this 3d printing world so there may be other solutions other than in Marlin.

ellensp, 2022-12-12:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-12-12 by ellensp

plundrigan, 2022-12-12:
Oh that is great. I had no idea. Thanks.