MarlinFirmware/Marlin #25087: Ability to have a Config file that can be read by Marlin on load so upgrades require less file comparision.

plundrigan, 2022-12-12:
### Is your feature request related to a problem? Please describe.

Each upgrade requires the full comparision of all files changed. While this is unavoidable in some cases. It would be nice to have a Marlin.cfg file that could be read at startup from SD card. 

### Describe the feature you want

The ability to turn on and off features at Marlin startup by reading a startup Marlin.cfg file. 

If the file is present on a SD card the file would be read and override the settings in the firmware itself. 

So for example, if I want to set BL_touch enabled. Instead of having to edit the Configuration.h file and taking out a // to uncomment.

I would be able to set a value on Marlin.cfg

BLTOUCH = true
or
X_PROBE_OFFSET_FROM_EXTRUDER =10

This would allow upgrades to go much faster, or for those that need to tweak settings often, it would minimize the need to recompile often. This is probably a big ask, but those with multiple machine (with slightly different configs)  this could be a big time saver.

closed 2022-12-12 by ellensp

ellensp, 2022-12-12:
Marlin is dynamic.
What does that mean? For eg when you compile with BLTOUCH disabled, the bltouch code is completely removed. These is no way to enable bltouch without compiling form source again.

This is the design philosophy of Marlin, anything that is not used is removed to save firmware space and reduce compile times. Any hardware change, Like adding a bltouch requires a recompile.

Also there is already something similar to this. Config.ini  used at compile time. To override Configuration settings

plundrigan, 2022-12-12:
I was not aware that that all the code was compiled into the firmware only the enabled sections. I will have to lookuo the config.ini you are referring too. I just thought if ALL the code/features could be compiled and a check was performed to look for a config file and if present enable/disable or set values. Then a compile would not require enabling and disabling features. The config would control that. However, if the there is a reason to keep the firmware as small as possible. I understand what you are saying. Thanks for the clarification.