MarlinFirmware/Marlin #25089: [BUG] (Error "You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN." even though they're not equal)

KeystrokeCascade, 2022-12-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When trying to compile Marlin for my SKR Mini E3 V2 it gives this error.  I have not touched any configuration files other than [`Configuration.h`](https://github.com/MarlinFirmware/Marlin/files/10208680/Configuration.txt), which doesn't even have definitions for `CONTROLLER_FAN_PIN` or `FAN_PIN`.

```
In file included from /home/keystroke/Documents/Programs/marlin/Marlin/src/inc/MarlinConfig.h:49:0,
                 from /home/keystroke/Documents/Programs/marlin/Marlin/src/HAL/shared/eeprom_if_i2c.cpp:28:
/home/keystroke/Documents/Programs/marlin/Marlin/src/inc/SanityCheck.h:2187:6: error: #error "You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN."
     #error "You cannot set CONTROLLER_FAN_PIN equal to FAN_PIN."
      ^
/home/keystroke/Documents/Programs/marlin/Marlin/src/HAL/shared/eeprom_if_i2c.cpp:35:28: fatal error: SlowSoftWire.h: No such file or directory
   #include <SlowSoftWire.h>
                            ^
compilation terminated.
exit status 1
Error compiling for board Generic STM32F103C series.
```

I've poked around in the pin files this board uses (`pins_BTT_SKR_MINI_E3_V2_0.h` and `pins_BTT_SKR_MINI_E3_common.h`) and found that `FAN_PIN` is defined as `PC6` while `CONTROLLER_FAN_PIN` is defined as `PC7`!

I've tried figuring out what I can but I only have limited C++ knowledge and this is my first time trying to set up 

### Bug Timeline

First time compiling for my SKR Mini E3 V2

### Expected behavior

It compiles.

### Actual behavior

It did not and gave the error detailed above.

### Steps to Reproduce

1. Download attached `Configuration.h` file
2. Try and compile for the SKR Mini E3 V2

### Version of Marlin Firmware

2.1.x

### Printer model

Da Vinci 1.0

### Electronics

BTT SKR Mini E3 V2.0

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-12-12:
> […] which doesn't even have definitions for `CONTROLLER_FAN_PIN` or `FAN_PIN`

Because `CONTROLLER_FAN_PIN` is defined in [`Configuration_adv.h`](https://github.com/MarlinFirmware/Marlin/blob/8a5ef7420be7b18f3692f4ef238718b9c3d64e8a/Marlin/Configuration_adv.h#L531-L556)

Please attach a copy in your next reply.

KeystrokeCascade, 2022-12-12:
As I said, I made no changes to any other files including the advanced config file.  As you helpfully linked, it has the global `CONTROLLER_FAN_PIN` definition commented out as is the case with mine.

thisiskeithb, 2022-12-12:
> Did you test the latest bugfix-2.1.x code?
> Yes, and the problem still exists.

> #define CONFIGURATION_H_VERSION 02010100

This is not current bugfix.

<details><summary>Your config compiles fine:</summary>

```prolog
Indexing .pio/build/STM32F103RC_btt/lib229/libTMCStepper.a
Linking .pio/build/STM32F103RC_btt/firmware.elf
Checking size .pio/build/STM32F103RC_btt/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.6% (used 10144 bytes from 49152 bytes)
Flash: [====      ]  44.2% (used 115892 bytes from 262144 bytes)
Building .pio/build/STM32F103RC_btt/firmware.bin
================================== [SUCCESS] Took 43.42 seconds ==================================

Environment      Status    Duration
---------------  --------  ------------
STM32F103RC_btt  SUCCESS   00:00:43.423
================================== 1 succeeded in 00:00:43.423 ==================================
 *  Terminal will be reused by tasks, press any key to close it.
```
</details>

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-12-12 by thisiskeithb

KeystrokeCascade, 2022-12-14:
Ok I figured out the error was caused by the target being the wrong sub-board (C instead of R series), however after fixing that and downloading some missing libraries ([SlowSoftI2CMaster](https://github.com/felias-fogg/SlowSoftI2CMaster), [SlowSoftWire](https://github.com/felias-fogg/SlowSoftWire) and TMCStepper) I'm now getting a bunch of errors relating to the SPI library and would really like to know what libraries you are using to successfully compile this, as the BTT docs and configuration don't say anything of the sort.

```
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/HAL_SPI.cpp: In function 'void spiRead(uint8_t*, uint16_t)':
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/HAL_SPI.cpp:127:54: error: call of overloaded 'dmaTransfer(int, uint8_t*, uint16_t&)' is ambiguous
   SPI.dmaTransfer(0, const_cast<uint8_t*>(buf), nbyte);
                                                      ^
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/HAL_SPI.cpp:127:54: note: candidates are:
In file included from /home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/HAL_SPI.cpp:33:0:
/home/keystroke/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:301:11: note: uint8 SPIClass::dmaTransfer(const void*, void*, uint16)
     uint8 dmaTransfer(const void * transmitBuf, void * receiveBuf, uint16 length);
           ^
/home/keystroke/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:302:11: note: uint8 SPIClass::dmaTransfer(uint16, void*, uint16)
     uint8 dmaTransfer(const uint16 value, void * receiveBuf, uint16 length);
           ^
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp: In function 'void waitSpiTxEnd(spi_dev*)':
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:97:20: error: redefinition of 'void waitSpiTxEnd(spi_dev*)'
 static inline void waitSpiTxEnd(spi_dev *spi_d) {
                    ^
In file included from /home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:34:0:
/home/keystroke/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:433:20: error: 'void waitSpiTxEnd(spi_dev*)' previously defined here
 static inline void waitSpiTxEnd(spi_dev *spi_d)
                    ^
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp: At global scope:
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:158:1: error: prototype for 'SPIClass::SPIClass(int8_t, int8_t, int8_t, int8_t)' does not match any in class 'SPIClass'
 SPIClass::SPIClass(int8_t mosi, int8_t miso, int8_t sclk, int8_t ssel) : SPIClass(1) {
 ^
In file included from /home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:34:0:
/home/keystroke/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:170:7: error: candidates are: constexpr SPIClass::SPIClass(SPIClass&&)
 class SPIClass {
       ^
/home/keystroke/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:170:7: error:                 constexpr SPIClass::SPIClass(const SPIClass&)
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:105:1: error:                 SPIClass::SPIClass(uint32_t)
 SPIClass::SPIClass(uint32_t spi_num) {
 ^
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:269:6: error: prototype for 'void SPIClass::beginTransaction(uint8_t, const SPISettings&)' does not match any in class 'SPIClass'
 void SPIClass::beginTransaction(uint8_t pin, const SPISettings &settings) {
      ^
In file included from /home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:34:0:
/home/keystroke/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:208:7: error: candidates are: void SPIClass::beginTransaction(uint8_t, SPISettings)
  void beginTransaction(uint8_t pin, SPISettings settings);
       ^
/home/keystroke/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:207:7: error:                 void SPIClass::beginTransaction(SPISettings)
  void beginTransaction(SPISettings settings) { beginTransaction(BOARD_SPI_DEFAULT_SS, settings); }
       ^
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:277:6: error: prototype for 'void SPIClass::beginTransactionSlave(const SPISettings&)' does not match any in class 'SPIClass'
 void SPIClass::beginTransactionSlave(const SPISettings &settings) {
      ^
In file included from /home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:34:0:
/home/keystroke/.arduino15/packages/stm32duino/hardware/STM32F1/2022.9.26/libraries/SPI/src/SPI.h:211:7: error: candidate is: void SPIClass::beginTransactionSlave(SPISettings)
  void beginTransactionSlave(SPISettings settings);
       ^
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp: In member function 'void SPIClass::dmaTransferSet(const void*, void*)':
/home/keystroke/Documents/programs/marlin/Marlin/src/HAL/STM32F1/SPI.cpp:386:20: error: 'ff' was not declared in this scope
     transmitBuf = &ff;
                    ^
exit status 1
Error compiling for board Generic STM32F103R series.
```

ellensp, 2022-12-14:
@KeystrokeCascade this is not a support forum

You should be using platformio not arduino ide. With Arduino ide you have to manually download boards and specific versions of libraries. But what version?     
Platform IO takes all this guess out out and automatically downloads what it needs,
Its also 10 * faster at compiles.

Get vscode with platformio https://marlinfw.org/docs/basics/install_platformio_vscode.html