MarlinFirmware/Marlin #25091: [BUG] Is PID_FAN_SCALING working/implemented?

MarkMan0, 2022-12-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

`PID_FAN_SCALING` doesn't seem to be working, and the `PIDRunner` doesn't use the Kf parameter of the PIDF struct. This lead me to believe PID_FAN_SCALING is not implemented into Marlin. Setting the Kf to high values (100) has no effect.

### Bug Timeline

12.12.2022

### Expected behavior

The compensation to work. Or when the Kf parameter is set too high, the temperature to overshoot the set value when turning on the fan

### Actual behavior

The kf parameter had no effect on the temperature 

### Steps to Reproduce

1. Enable PID_FAN_SCALING 
2. Observe temperature graph with Kf set to 0
3. Observe temperature with Kf set to 100

### Version of Marlin Firmware

Marlin bugfix-2.1.x (Dec 12 2022 21:15:52) 

### Printer model

Ender 3

### Electronics

SKR v1.3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10211748/Configuration.zip)

ellensp, 2022-12-13:
It looks like this feature was lost in https://github.com/MarlinFirmware/Marlin/pull/24389 from Marlin/src/module/temperature.cpp
PID_FAN_SCALING just vanished...

MarkMan0, 2022-12-13:
I'll try to reimplement it then. Guess the cleanest way would be to add a `calculate()` method to `PID_t, PIDC_t, PIDF_t, PIDCF_t`. Or was there a specific reason these structs don't have methods?

Btw is `PID_EXTRUSION_SCALING` working? Because the contant for it is in the PID struct too.

thisiskeithb, 2022-12-13:
Closing since PR https://github.com/MarlinFirmware/Marlin/pull/25096 has been opened.

closed 2022-12-13 by thisiskeithb