MarlinFirmware/Marlin #25098: ui_320x240.cpp

SuperSpooner, 2022-12-15:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The rectangle around the XYZ axis positions on the TFT screen with resolution 320x240 is calling the wrong colour.
Box used to be white but is now orange.

ui_320x240.cpp calls for:
tft.add_rectangle(0, 0, 312, 24, COLOR_AXIS_HOMED);

When I believe it should be:
tft.add_rectangle(0, 0, 312, 24, COLOR_AXIS_FRAME);

### Bug Timeline

Noticed when changing to 2.1.1. Not present in 2.0.x bugfix 2022-05-10

### Expected behavior

Rectangle color should always match COLOR_AXIS_FRAME

### Actual behavior

Rectangle color is coded to match COLOR_AXIS_HOMED

### Steps to Reproduce

Compile and use 320x240 touchscreen

### Version of Marlin Firmware

2.1.1

### Printer model

Anet ET4

### Electronics

Stock

### Add-ons

None

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Simplify3D

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin 2.1.1 ET4 Configs.zip](https://github.com/MarlinFirmware/Marlin/files/10239759/Marlin.2.1.1.ET4.Configs.zip)

jmz52, 2022-12-16:
Actually this is an intended behavior - when axis are homed text and frame are of the same color.
And it is was implemented this way from the very beginning
https://github.com/jmz52/Marlin/blob/tft_and_touch_screen/Marlin/src/lcd/tft/ui_320x240.cpp#L258

And default value for COLOR_AXIS_HOMED is COLOR_WHITE
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/lcd/tft/tft_color.h#L105

thisiskeithb, 2022-12-16:
You're also using a custom theme, so colors will be orange:

```cpp
/* ET4/ET5 Black Theme */

#define COLOR_RED2              0xF003 // #F70019
#define COLOR_ORANGE2           0xFDE0 // #FFBE00
#define COLOR_LIME2             0xA7E0 // #A5FF00
#define COLOR_BLACK2            0x2124 // #212421

#define COLOR_BACKGROUND        COLOR_BLACK2
#define COLOR_SELECTION_BG      COLOR_ORANGE2
#define COLOR_COLD              COLOR_WHITE
#define COLOR_HOTEND            COLOR_RED2
#define COLOR_HEATED_BED        COLOR_RED2
#define COLOR_CHAMBER           COLOR_WHITE
#define COLOR_FAN               COLOR_WHITE
#define COLOR_AXIS_FRAME        COLOR_WHITE
#define COLOR_AXIS_HOMED        COLOR_ORANGE2
#define COLOR_AXIS_NOT_HOMED    COLOR_ORANGE2
#define COLOR_RATE_100          COLOR_LIME2
#define COLOR_RATE_ALTERED      COLOR_ORANGE2
#define COLOR_PRINT_TIME        COLOR_WHITE
#define COLOR_PROGRESS_BAR      COLOR_ORANGE2
#define COLOR_STATUS_MESSAGE    COLOR_ORANGE2
#define COLOR_SD_ENABLED        COLOR_CONTROL_ENABLED
#define COLOR_SD_DISABLED       COLOR_CONTROL_DISABLED
#define COLOR_MENU_TEXT         COLOR_WHITE
#define COLOR_MENU_VALUE_FONT   COLOR_ORANGE2
#define COLOR_SLIDER            COLOR_ORANGE2
#define COLOR_INCREASE          COLOR_WHITE
#define COLOR_DECREASE          COLOR_WHITE
#define COLOR_TICK              COLOR_VIVID_GREEN
```

closed 2022-12-16 by thisiskeithb