MarlinFirmware/Marlin #25102: [BUG] host loses USB connection on SD insertion (MKS Robin Nano 1.2, MKS TFT 3.5)

dchauran, 2022-12-16:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When I insert the SD card, I frequently see octoprint lose connection to the printer. dmesg says the device disconnected:

[986913.464940] ch341-uart ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
[986913.465452] ch341-uart ttyUSB1: usb_serial_generic_read_bulk_callback - urb stopped: -32
[986913.483590] usb 1-1.3: USB disconnect, device number 31
[986913.484052] usb 1-1.3: failed to send control message: -19
[986913.484315] ch341-uart ttyUSB1: ch341-uart converter now disconnected from ttyUSB1
[986913.484355] ch341 1-1.3:1.0: device disconnected
[986913.827982] usb 1-1.3: new full-speed USB device number 32 using xhci_hcd
[986913.964011] usb 1-1.3: New USB device found, idVendor=1a86, idProduct=7523, bcdDevice= 2.64
[986913.964022] usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[986913.964029] usb 1-1.3: Product: USB Serial
[986913.966625] ch341 1-1.3:1.0: ch341-uart converter detected
[986913.972127] usb 1-1.3: ch341-uart converter now attached to ttyUSB1

In general, since updating from bugfix 2.0 to 2.1, USB seems very unstable, and often disconnects with the same errors, but this is the only thing that I can reproduce reliably.

### Bug Timeline

After updating from 2.0.x to 2.1.x

### Expected behavior

I insert SD card and USB stays connected.

### Actual behavior

Upon inserting the SD card, octoprint loses the serial connection.

### Version of Marlin Firmware

Bugfix 2.1.x, 2022-12-13

### Printer model

Pyramid A1.1, modified

### Electronics

Stock board+tft, tmc 2208 standalone throughout, creality sprite extruder

### Add-ons

MKS TFT, Octopi, MKS Wifi module

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

My current configuration is always kept up to date here, as are minor customizations to the firmware (note that I have an extra config file, as I was at one point maintaining firmware for the stock version as well as my customized version):

https://github.com/dchauran/marlin

thinkyhead, 2022-12-18:
Would you be able to test some `bugfix-2.0.x` commits to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke. This technique can be facilitated by using the `git bisect` command from the command-line.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

dchauran, 2022-12-18:
I can try, though that may prove troublesome with config file differences, but I actually have a branch for this printer that is still on 2.0.x from about 2000 commits ago, albeit with a slightly different config, but it should work for testing.

closed 2023-02-28 by github-actions[bot]