MarlinFirmware/Marlin #25105: [BUG] (generic delta powerloss recovery doesn't work)

Enzo-Enz, 2022-12-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

after updating my delta printer firmware,the powerloss recovery doesn't work
I also enabled #define SAVE_INFO_INTERVAL_MS 0

### Bug Timeline

new

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

diy delta

### Electronics

skr pro v1.2

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/10251928/config.zip)

Foxies-CSTL, 2024-04-02:
Here's my contribution to the problems encountered, as I have users who use it.

The function no longer works with version 2.1.2.2: no "PLR" file creation/writing when the function in the menu is activated.

Conditions of use :

- The name of the file to be printed is identical to that contained in the GCode.

- SD Class10 card,

- Name less than 24 characters,

- file extension ".gcodes" or ".gco",

- No connection with a host (Octoprint, etc)

Several attempts, one of which succeeded when I activated the "PowerLoss" function in the menu when preparing the heating of the file to be printed.

But the recovery sequence was catastrophic in the positioning of the print head: a "Home" then a positioning at mid-height at the end of the bed to return to the position for resuming printing. I tried to reproduce this test for a data recording but the reprsse movements were not identical and the head crashed at the end of the bed. 
Tests carried out with a Delta. This function worked fine in 2.0.9!
Same behavior with Bump 24/03/2024 of V2.0.3.
Also related to #26273

thisiskeithb, 2024-04-02:
> Tests carried out with a Delta. This function worked fine in 2.0.9!

I renamed this back to a bug report since it worked in prior versions.

Can you add `#define DEBUG_POWER_LOSS_RECOVERY` to your config and log the serial output? It might give more clues as to why it no longer works on deltas.

Foxies-CSTL, 2024-04-02:
> Can you add #define DEBUG_POWER_LOSS_RECOVERY to your config and log the serial output?

It's difficult to register the serial port (debug) with the Octoprint terminal because the loss of power disconnects the link and resuming printing by reconnecting Octoprint is random. What's more, I managed to create a new PLR file by switching off the power supply at 50% of the print run, but printing resumes in a different way!... it resumes printing from the beginning and not at the switch-off point.

[Log-BugPowerLoss.txt](https://github.com/MarlinFirmware/Marlin/files/14844261/Log-BugPowerLoss.txt)

Foxies-CSTL, 2024-04-02:
Here's another one where the print head has moved to the outer edge of the bed=> emergency stop :-(
[Log-BugPowerLoss2.txt](https://github.com/MarlinFirmware/Marlin/files/14844440/Log-BugPowerLoss2.txt)

closed 2024-07-17 by github-actions[bot]