MarlinFirmware/Marlin #25106: Artillery GPLv3 license violation - did not release source code

Ltek, 2022-12-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Artillery 3D Printers is distributing an update to the original version they release source code to well over a year ago. They have been asked several times by several people to post source code and its been over a month, they have not... and never answer the requestors. Please force them to release the source code. Thank you.

Binaries released Nov 30, 2021... https://drive.google.com/file/d/1nGygGYUGs4zX6pLU1wPiTxzZ1-vHOm6W/view
Source code for above... https://github.com/artillery3d/sidewinder-x2-firmware

Binaries they released Nov 9, 2022...  https://drive.google.com/file/d/1el6oaaCBEMHrdP1kv8SIyJXAafAoO8I0/view

### Expected behavior

Source Code posted publicly for latest Marlin binaries posted publicly

### Actual behavior

no source code is posted for binaries posted Nov 9, 2022.

### Version of Marlin Firmware

2.1.x

### Printer model

Artillery Sidewinder X2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

n/a

ellensp, 2022-12-18:
A quote from the head Marlin developer.

"Thinkyhead — 12/27/2021 7:06 PM
Clarifying: Under GPL the source code does not need to be made public even for publicly-posted binaries. But it must be provided to anyone who has the binary and requests the source code. They cannot be denied without violating the terms. And, the recipient can do whatever they want with the source, including making it public."

So your 
"Expected behavior: Source Code posted publicly for latest Marlin binaries posted publicly" is not a requirement of GPL

But their refusing to provide source to those with the binary on request is a violation.

The easiest way is of course to publicly release it, then you don't get lots of people asking for it.

FYI https://github.com/MarlinFirmware/Marlin/issues/24801
You should get them added to this list https://www.th3dstudio.com/3d-printer-gpl-violation-list/

Other that this there is nothing more "we" can do.

closed 2022-12-18 by ellensp

Ltek, 2022-12-18:
I posted a request on their Facebook page in the thread it was announce in, the day it was announced, and again yesterday...
https://www.facebook.com/groups/artilleryswx1/permalink/1161634197794577

Today I sent them a FB messenger message and email request to support@artillery3d.com
and opened a support ticket in their system