MarlinFirmware/Marlin #25107: [BUG] Ender-3 wheel & button stopped working with 2.1.2

vpont, 2022-12-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I just flashed 2.1.2 to my Ender-3 (CrealityV1) and the wheel and button from the control panel stopped working.
Flashing back 2.1.1 makes it work again.

### Expected behavior

I expect it to work.

### Actual behavior

Unable to select with wheel or click.

### Version of Marlin Firmware

2.1.2

### Printer model

Creality Ender-3 (CrealityV1)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I'm using [default configs](https://github.com/MarlinFirmware/Configurations/tree/2.1.2/config/examples/Creality/Ender-3/CrealityV1).

ellensp, 2022-12-18:
Encoder pins are not being defined

2.1.1
PIN:  10   Port: D2 (A21)  BTN_EN2                                Input  = 1  
PIN:  11   Port: D3 (A20)  BTN_EN1                                Input  = 1  
PIN:  16   Port: C0 (A15)  BTN_ENC                                Input  = 1  

2.1.2
PIN:  10   Port: D2 (A21)  <unused/unknown>                       Input  = 0                         
PIN:  11   Port: D3 (A20)  <unused/unknown>                       Input  = 0                         
PIN:  16   Port: C0 (A15)  <unused/unknown>                       Input  = 0                         

Issue is due to LCD_PINS_DEFINED
This tells pins_SANGUINOLOLU_11.h not to set LCD pins, but pins_MELZI_CREALITY.h only set the spi pins, not the encoder pins

Broken in https://github.com/MarlinFirmware/Marlin/pull/25012 after thinkyhead modified the PR

ellensp, 2022-12-18:
This should fix it, please test  

```DIFF
diff --git a/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h b/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h
index 76478dd1a0..4301ffa578 100644
--- a/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h
+++ b/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h
@@ -50,6 +50,9 @@
   #define LCD_PINS_RS                         28  // ST9720 CS
   #define LCD_PINS_ENABLE                     17  // ST9720 DAT
   #define LCD_PINS_D4                         30  // ST9720 CLK
+  #define BTN_ENC                             16
+  #define BTN_EN1                             11
+  #define BTN_EN2                             10
 
   #define LCD_PINS_DEFINED
 #endif

```

vpont, 2022-12-18:
Confirmed working again with the provided patch 👍

thisiskeithb, 2022-12-18:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/0cf2443aa0e816c34b73fc598cd160852b63b132 and backported to `2.1.2`, so [grab a fresh copy](https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.1.2.zip).

thinkyhead, 2022-12-18:
Patch patched! Thanks for the report.

closed 2022-12-18 by thinkyhead