MarlinFirmware/Marlin #25109: [BUG] 2.1.2 Release build fails using Anycubic Chiron example configs

SwiftNick, 2022-12-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Using the example config files for the AnyCubic Chiron, compilation fails due to missing LCD_SERIAL_PORT definition.

The previous definition in configuration.h has been removed  [2.1.1 config file](https://github.com/MarlinFirmware/Configurations/blob/f52c85b0e3b241f8fabb4a902a4aa53775fdeb93/config/examples/AnyCubic/Chiron/Configuration.h#L2990)
There is some refactoring around the DGUS_LCD code which seems to relate to this.

In configuration.h, restoring LCD_SERIAL_PORT 3 or adding HAS_DGUS_LCD 1 to the code block resolves the issue.

```
#define ANYCUBIC_LCD_CHIRON
#if EITHER(ANYCUBIC_LCD_I3MEGA, ANYCUBIC_LCD_CHIRON)
  #define HAS_DGUS_LCD 1
```  
  @thinkyhead  not sure which you would prefer as I don't know the extent of your refactoring.
  

### Bug Timeline

Introduced in 2.1.2 release

### Expected behavior

Code should build without error

### Actual behavior

Compilation fails with sanity check error

### Version of Marlin Firmware

2.1.2

### Printer model

Anycubic Chiron

### Electronics

stock build

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Nuck-TH, 2022-12-18:
Chiron screen with "white" interface is not DGUS, so that define should not be automatically activated.

SwiftNick, 2022-12-18:
> Chiron screen with "white" interface is not DGUS, so that define should not be automatically activated.

So the LCD_SERIAL_PORT 3 needs to be reinstated or the ANYCUBIC_LCD_CHIRON needs to be included in the automatic port definitions in https://github.com/MarlinFirmware/Marlin/blob/a7eacbcc49dc74c6e49bae7d82f726c801eaf3fd/Marlin/src/inc/Conditionals_adv.h#L1079

```
#ifndef LCD_SERIAL_PORT
  #if HAS_DWIN_E3V2 || IS_DWIN_MARLINUI || HAS_DGUS_LCD
    #if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_MINI_E3_V3_0, BTT_SKR_E3_TURBO, BTT_OCTOPUS_V1_1)
      #define LCD_SERIAL_PORT 1
    #elif MB(CREALITY_V24S1_301, CREALITY_V24S1_301F4, CREALITY_V423, MKS_ROBIN)
      #define LCD_SERIAL_PORT 2 // Creality Ender3S1, MKS Robin
    #else
      #define LCD_SERIAL_PORT 3 // Other boards
```

SwiftNick, 2022-12-18:
So the LCD_SERIAL_PORT 3 needs to be reinstated or the ANYCUBIC_LCD_CHIRON needs to be included in the automatic port definitions in https://github.com/MarlinFirmware/Marlin/blob/a7eacbcc49dc74c6e49bae7d82f726c801eaf3fd/Marlin/src/inc/Conditionals_adv.h#L1079

```
#ifndef LCD_SERIAL_PORT
  #if HAS_DWIN_E3V2 || IS_DWIN_MARLINUI || HAS_DGUS_LCD
    #if MB(BTT_SKR_MINI_E3_V1_0, BTT_SKR_MINI_E3_V1_2, BTT_SKR_MINI_E3_V2_0, BTT_SKR_MINI_E3_V3_0, BTT_SKR_E3_TURBO, BTT_OCTOPUS_V1_1)
      #define LCD_SERIAL_PORT 1
    #elif MB(CREALITY_V24S1_301, CREALITY_V24S1_301F4, CREALITY_V423, MKS_ROBIN)
      #define LCD_SERIAL_PORT 2 // Creality Ender3S1, MKS Robin
    #else
      #define LCD_SERIAL_PORT 3 // Other boards
```

thinkyhead, 2023-02-08:
I should set it in the example configuration, since the Chiron's LCD can be hooked up to any serial port on any board. A default of 3 is possibly the only option when the Trigorilla board is selected, so maybe it makes sense to assign it in the Trigorilla pins file when the Anycubic serial LCD is selected, in prparation, as there is a major refactor of LCD pin assignments on the horizon.

SwiftNick, 2023-04-15:
Bump
just retested using bugfix-2.1.x & configurations bugfix-2.1.x the LCD_SERIAL_PORT definition is still missing.

thisiskeithb, 2023-04-15:
> I should set it in the example configuration, since the Chiron's LCD can be hooked up to any serial port on any board

Transferring to the Configurations repo…