MarlinFirmware/Marlin #25110: I can't get Marlin to compile

bob5731, 2022-12-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

PS C:\Marlin-1> platformio run --silent -e STM32G0B1RE_btt
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   76 |   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:81:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   81 |   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:650:4: error: #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
  650 |   #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   76 |   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:81:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   81 |   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:650:4: error: #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
  650 |   #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   76 |   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:81:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   81 |   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:650:4: error: #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
  650 |   #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   76 |   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:81:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   81 |   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
c:\marlin-1\marlin\src\inc\SanityCheck.h:650:4: error: #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
  650 |   #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
      |    ^~~~~
In file included from c:\marlin-1\marlin\src\inc\marlinconfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_bl24cxx.cpp:32:
c:\marlin-1\marlin\src\inc\SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
[... 178 lines cut ...]
### Add-ons

BTT SKR Mini E3 V3.0

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Cura

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I need BLTouch and runout and PROBE_OFFSET_WIZARD..
I add Configuration.h and Configuration_adv.h
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10254524/Marlin.zip)

EvilGremlin, 2022-12-18:
user error, it tells you what to do in verty first message

bob5731, 2022-12-18:
I got the Configuration files from the Marlin 2.1.1 release.

EvilGremlin, 2022-12-18:
No you don't - configs are 2.1.0

bob5731, 2022-12-18:
[https://github.com/MarlinFirmware/Configurations/tree/release-2.1.1](url)

EvilGremlin, 2022-12-18:
compare CONFIGURATION_H_VERSION

bob5731, 2022-12-18:
CONFIGURATION_H_VERSION code

/**
 * Marlin 3D Printer Firmware
 * Copyright (c) 2020 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 *
 * Based on Sprinter and grbl.
 * Copyright (c) 2011 Camiel Gubbels / Erik van der Zalm
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <https://www.gnu.org/licenses/>.
 *
 */
#pragma once

////////////////////////////
// VENDOR VERSION EXAMPLE //
////////////////////////////

/**
 * Marlin release version identifier
 */
//#define SHORT_BUILD_VERSION "bugfix-2.1.x"

/**
 * Verbose version identifier which should contain a reference to the location
 * from where the binary was downloaded or the source code was compiled.
 */
//#define DETAILED_BUILD_VERSION SHORT_BUILD_VERSION

/**
 * The STRING_DISTRIBUTION_DATE represents when the binary file was built,
 * here we define this default string as the date where the latest release
 * version was tagged.
 */
//#define STRING_DISTRIBUTION_DATE "2022-12-18"

/**
 * Defines a generic printer name to be output to the LCD after booting Marlin.
 */
//#define MACHINE_NAME "3D Printer"

/**
 * The SOURCE_CODE_URL is the location where users will find the Marlin Source
 * Code which is installed on the device. In most cases —unless the manufacturer
 * has a distinct Github fork— the Source Code URL should just be the main
 * Marlin repository.
 */
//#define SOURCE_CODE_URL "github.com/MarlinFirmware/Marlin"

/**
 * Default generic printer UUID.
 */
//#define DEFAULT_MACHINE_UUID "cede2a2f-41a2-4748-9b12-c55c62f367ff"

/**
 * The WEBSITE_URL is the location where users can get more information such as
 * documentation about a specific Marlin release.
 */
//#define WEBSITE_URL "marlinfw.org"

/**
 * Set the vendor info the serial USB interface, if changable
 * Currently only supported by DUE platform
 */
//#define USB_DEVICE_VENDOR_ID           0x0000
//#define USB_DEVICE_PRODUCT_ID          0x0000
//#define USB_DEVICE_MANUFACTURE_NAME    WEBSITE_URL

EvilGremlin, 2022-12-18:
Please pay more attention to what you download.
your configs are CONFIGURATION_H_VERSION 02010000
2.1.1 is CONFIGURATION_H_VERSION 02010100
now you post version.h of bugfix-2.1.x which have CONFIGURATION_H_VERSION 02010200

bob5731, 2022-12-18:
I cloned the Marlin 2.1.1 repo

EvilGremlin, 2022-12-18:
Repo is one for everything and default branch is bugfix-2.1.x. You can't clone 2.1.1, you must checkout to 2.1.1 tag after cloning.

bob5731, 2022-12-18:
my bad 2.1.2

bob5731, 2022-12-18:
I redownload MarlinFirmware 2.1.2 and recompiled.
now get

S C:\Marlin-2.1.2> platformio run --silent -e STM32G0B1RE_btt
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
   76 |   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:81:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   81 |   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:650:4: error: #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
  650 |   #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."     
   76 |   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:81:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   81 |   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:650:4: error: #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
  650 |   #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."     
   76 |   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:81:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   81 |   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:650:4: error: #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
  650 |   #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_bl24cxx.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:76:4: error: #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."     
   76 |   #error "Your Configuration.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration.h."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_bl24cxx.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:81:4: error: #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
   81 |   #error "Your Configuration_adv.h file is for an old version of Marlin. Downgrade Marlin or upgrade your Configuration_adv.h."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\eeprom_bl24cxx.cpp:32:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:650:4: error: #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
  650 |   #error "LCD_SET_PROGRESS_MANUALLY is now SET_PROGRESS_MANUALLY."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
[... cut ...]
      |                                             ^
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\eeprom_if_iic.cpp.o] Error 1
============================== [FAILED] Took 30.10 seconds ==============================

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  FAILED    00:00:30.102
========================= 1 failed, 0 succeeded in 00:00:30.102 ========================= 
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"

EvilGremlin, 2022-12-18:
same: download approprite version of configs
release-2.1.2 branch in configs repo

bob5731, 2022-12-18:
I jest update the configs. now waiting on it to compile . it compile whit no errors.
Now need to get BLTouch and runout and PROBE_OFFSET_WIZARD working!!
How do I setup BLTouch and runout and PROBE_OFFSET_WIZARD?

thisiskeithb, 2022-12-18:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-12-18 by thisiskeithb

bob5731, 2022-12-19:
now win I setup AUTO_BED_LEVELING_BILINEAR I get
l'm setup BLTouch.

PS C:\Marlin-2.1.2> platformio run --silent -e STM32G0B1RE_btt
Marlin\src\inc\Warnings.cpp:731:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  731 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --target clean --silent -e STM32G0B1RE_btt
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --silent -e STM32G0B1RE_btt
Marlin\src\inc\Warnings.cpp:731:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  731 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --target clean --silent -e STM32G0B1RE_btt
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --silent -e STM32G0B1RE_btt
Marlin\src\inc\Warnings.cpp:731:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  731 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --target clean --silent -e STM32G0B1RE_btt
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --silent -e STM32G0B1RE_btt
Marlin\src\inc\Warnings.cpp:731:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  731 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --target clean --silent -e STM32G0B1RE_btt
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --silent -e STM32G0B1RE_btt
Marlin\src\inc\Warnings.cpp:731:6: warning: #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)" [-Wcpp]
  731 |     #warning "Motherboard DIAG jumpers must be removed when SENSORLESS_HOMING is disabled. (Define DIAG_JUMPERS_REMOVED to suppress this warning.)"
      |      ^~~~~~~
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --target clean --silent -e STM32G0B1RE_btt
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --silent -e STM32G0B1RE_btt
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1975:4: error: #error "Select only one of: MESH_BED_LEVELING, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL."
 1975 |   #error "Select only one of: MESH_BED_LEVELING, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1975:4: error: #error "Select only one of: MESH_BED_LEVELING, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL."
 1975 |   #error "Select only one of: MESH_BED_LEVELING, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\Servo.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1975:4: error: #error "Select only one of: MESH_BED_LEVELING, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL."
 1975 |   #error "Select only one of: MESH_BED_LEVELING, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL."
      |    ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:27:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1975:4: error: #error "Select only one of: MESH_BED_LEVELING, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL."
 1975 |   #error "Select only one of: MESH_BED_LEVELING, AUTO_BED_LEVELING_LINEAR, AUTO_BED_LEVELING_3POINT, AUTO_BED_LEVELING_BILINEAR or AUTO_BED_LEVELING_UBL."
      |    ^~~~~
[... cut ...]
*** [.pio\build\STM32G0B1RE_btt\src\src\HAL\STM32\eeprom_if_iic.cpp.o] Error 1
============================== [FAILED] Took 25.45 seconds ==============================

Environment      Status    Duration
---------------  --------  ------------
STM32G0B1RE_btt  FAILED    00:00:25.446
========================= 1 failed, 0 succeeded in 00:00:25.446 ========================= 
PS C:\Marlin-2.1.2> echo "done" >"C:\Users\v\AppData\Local\Temp\ipc"
PS C:\Marlin-2.1.2> platformio run --silent -e STM32G0B1RE_btt

thisiskeithb, 2022-12-19:
@bob5731 This is not a support forum. The sanity check is telling you what to do.

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

bob5731, 2022-12-21:
can me one help me yut?

EvilGremlin, 2022-12-21:
You were answered, and errors tell you what to do very clearly. if you have other questions - go to Discord or something.