MarlinFirmware/Marlin #25111: [BUG] Can't set input shaping frequency

040medien, 2022-12-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I've activated input shaping in 2.1.2, but whenever I try to set the frequency on either one axis or both, I'm getting this error message, except if I set frequency to 0:

```
Send: M593 X F18.4
Recv: ?Frequency (F) must be greater than 0.00 or 0 to disable
```

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10254657/Configuration.zip)

### Bug Timeline

New bug

### Expected behavior

I expect that I can set the input shaping frequency as documented.

### Actual behavior

Frequency doesn't get set, nonsensical error returned.

### Version of Marlin Firmware

2.1.2

### Printer model

Wanhao Duplicator 6

### Electronics

Bigtreetech-Skr-V1.3

### Add-ons

TMC2209

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

040medien, 2022-12-18:
So interestingly it looks like max_freq somehow is getting evaluated as 0

thisiskeithb, 2022-12-18:
Fixed in https://github.com/MarlinFirmware/Marlin/commit/ebcf26ec5daa82eb165afe68058a8143fb58208a and backported to `2.1.2`, so [grab a fresh copy](https://github.com/MarlinFirmware/Marlin/archive/refs/tags/2.1.2.zip).

closed 2022-12-18 by thisiskeithb