MarlinFirmware/Marlin #25115: [BUG] Enabling both MPCTEMP and ADAPTIVE_FAN_SLOWING breaks build

LMBernardo, 2022-12-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

### The Issue

Building fails when enabling both MPCTEMP and ADAPTIVE_FAN_SLOWING. This combination is not mentioned anywhere in the [MPCTEMP Documentation](https://marlinfw.org/docs/features/model_predictive_control.html), nor is it mentioned anywhere in the config files, and it seems like these features should be compatible.

### Possible Fix

Looking at `temperature.cpp` in [the original PR](https://github.com/MarlinFirmware/Marlin/pull/13066/files#diff-9c1d09e39e0d06d83cd4a44f54ff41e1cc364477c4dc39ecacf106a9b75540b7R99-R402), line 99 reads:

`#if ENABLED(NO_FAN_SLOWING_IN_PID_TUNING)`

It looks like this is a mistake, particularly when comparing to line 399, and should instead be:

`#if ENABLED(ADAPTIVE_FAN_SLOWING)`

I believe that this has gone unnoticed for so long because MPCTEMP is still noted as experimental and unreliable, and I don't think ADAPTIVE_FAN_SLOWING is a particularly common option to enable either. Most likely related to #24687.

### Bug Timeline

Old - this probably existed since the original PR on Feb 4, 2019

### Expected behavior

Marlin should either build successfully, or fail with a message stating that these two options cannot be used together. If the latter, this should also be documented somewhere.

### Actual behavior

Build fails with the error:

```
Marlin\src\module\temperature.cpp: In member function 'void Temperature::tr_state_machine_t::run(const_celsius_float_t, const_celsius_float_t, heater_id_t, uint16_t, celsius_t)':
Marlin\src\module\temperature.cpp:2872:15: error: 'adaptive_fan_slowing' was not declared in this scope
           if (adaptive_fan_slowing && heater_id >= 0) {
               ^~~~~~~~~~~~~~~~~~~~
*** [.pio\build\mega2560\src\src\module\temperature.cpp.o] Error 1
```

### Steps to Reproduce

1. Checkout most the recent Marlin bugfix-2.1.x code @ [fb5a595](https://github.com/MarlinFirmware/Marlin/commit/fb5a595b348824a4fe0adca6dfc796105c25e0cd)

2. Disable PIDTEMP and enable MPCTEMP in Configuration.h

3. Enable ADAPTIVE_FAN_SLOWING in Configuration_adv.h

4. Try to build

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

N/A

### Electronics

N/A

### Add-ons

N/A

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/10261355/Config.zip)

thisiskeithb, 2022-12-19:
> AFAIK MPC should take care of fan and then adaptive fan shouldn't be necessary.

With `MPC_INCLUDE_FAN` enabled, `ADAPTIVE_FAN_SLOWING` shouldn't be necessary, but it's not specifically sanity checked.

tombrazier, 2023-01-28:
I have always had ADAPTIVE_FAN_SLOWING enabled along with MPC and no compile error.

thisiskeithb, 2023-01-28:
> I have always had ADAPTIVE_FAN_SLOWING enabled along with MPC and no compile error.

I would guess something broke it recently.

Using a fresh `bugfix-2.1.x` pull & disabling `PIDTEMP`, enabling `MPCTEMP` & `ADAPTIVE_FAN_SLOWING` produces the error:
```prolog
Indexing .pio/build/mega2560/libFrameworkArduino.a
Marlin/src/module/temperature.cpp: In member function 'void Temperature::tr_state_machine_t::run(const_celsius_float_t, const_celsius_float_t, heater_id_t, uint16_t, celsius_t)':
Marlin/src/module/temperature.cpp:2893:15: error: 'adaptive_fan_slowing' was not declared in this scope
           if (adaptive_fan_slowing && heater_id >= 0) {
               ^~~~~~~~~~~~~~~~~~~~
*** [.pio/build/mega2560/src/src/module/temperature.cpp.o] Error 1
```

tombrazier, 2023-01-28:
I do not get the compile error because I have PIDTEMPBED defined. The problem is not with enabling MPCTEMP, it is with disabling PIDTEMP. The same happens if you disable PIDTEMP for bang bang control.

The following code incorrectly couples ADAPTIVE_FAN_SLOWING with PID.
https://github.com/MarlinFirmware/Marlin/blob/67bc855000345921281a2c8c210e4363542f6a81/Marlin/src/module/temperature.h#L1176-L1180

tombrazier, 2023-01-28:
I think https://github.com/tombrazier/Marlin/tree/25115 should fix it. Haven;t tried it though.

thisiskeithb, 2023-01-29:
Closing since there's now a PR fix this:
- https://github.com/MarlinFirmware/Marlin/pull/25314

closed 2023-01-29 by thisiskeithb