MarlinFirmware/Marlin #25116: [BUG] ESP32 compilation errors

mtk11, 2022-12-19:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Trying to compile Marlin for ESP32, but getting lots of errors:

### Expected behavior

compile with no errors

### Actual behavior

```

C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp: In function 'adc1_channel_t get_channel(int)':
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: error: 'ADC1_GPIO39_CHANNEL' was not declared in this scope
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:216:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 39: return ADC1_CHANNEL(39);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: note: suggested alternative: 'ADC1_CHANNEL'
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:216:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 39: return ADC1_CHANNEL(39);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: error: 'ADC1_GPIO36_CHANNEL' was not declared in this scope
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:217:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 36: return ADC1_CHANNEL(36);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: note: suggested alternative: 'ADC1_CHANNEL'
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:217:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 36: return ADC1_CHANNEL(36);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: error: 'ADC1_GPIO35_CHANNEL' was not declared in this scope
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:218:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 35: return ADC1_CHANNEL(35);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: note: suggested alternative: 'ADC1_CHANNEL'
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:218:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 35: return ADC1_CHANNEL(35);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: error: 'ADC1_GPIO34_CHANNEL' was not declared in this scope
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:219:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 34: return ADC1_CHANNEL(34);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: note: suggested alternative: 'ADC1_CHANNEL'
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:219:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 34: return ADC1_CHANNEL(34);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: error: 'ADC1_GPIO33_CHANNEL' was not declared in this scope
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:220:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 33: return ADC1_CHANNEL(33);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: note: suggested alternative: 'ADC1_CHANNEL'
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:220:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 33: return ADC1_CHANNEL(33);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: error: 'ADC1_GPIO32_CHANNEL' was not declared in this scope
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:221:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 32: return ADC1_CHANNEL(32);
                     ^~~~~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:212:27: note: suggested alternative: 'ADC1_CHANNEL'
 #define ADC1_CHANNEL(pin) ADC1_GPIO ## pin ## _CHANNEL
                           ^~~~~~~~~
C:\Marlin\Marlin-bugfix-2.1.x\Marlin\src\HAL\ESP32\HAL.cpp:221:21: note: in expansion of macro 'ADC1_CHANNEL'
     case 32: return ADC1_CHANNEL(32);
                     ^~~~~~~~~~~~
exit status 1
Error compiling for board ESP32 Dev Module.

```

### Version of Marlin Firmware

Marlin-bugfix-2.1.x

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

Default config, just ESP32 target board.

[Marlin_config.zip](https://github.com/MarlinFirmware/Marlin/files/10263190/Marlin_config.zip)

ellensp, 2022-12-19:
Provided configs complies without issues

That define is part of the framework-arduinoespressif32
in .platformio/packages/framework-arduinoespressif32/tools/sdk/include/soc/soc/adc_channel.h:#define ADC1_GPIO39_CHANNEL     ADC1_CHANNEL_3

You have a corrupt platformio install

Delete the .platformio directory in you user home directory and restart vscode.
Give vscode lots of time to download and reinstall platformio 

Then hit build and wait patiently for it to download and install framework-arduinoespressif32 (and other required libraries) 
Do not close vscode or restart your machine if it looks like it is not doing anything, give it time.

closed 2022-12-19 by ellensp

mtk11, 2022-12-20:
I am using Arduino IDE 1.8.19 to compile Marlin.

EvilGremlin, 2022-12-20:
This is your error, don't use arduino.

mtk11, 2022-12-20:
I like Arduino IDE much more than Visual Studio Code. Since there is a Marlin.ino file this issue suppose to be fixed.

EvilGremlin, 2022-12-20:
No, this is not an isuue and it won't be fixed. Arduino not supposed to work for anything except AVR, and even then in limited capability. You must use vscode+platformio for any other architecture.

mtk11, 2022-12-20:
Until some time ago, Marlin had compiled fine under Arduino IDE.

EvilGremlin, 2022-12-20:
Coincidence. If you look in ini you'll see pinned espressif32 framework version, it's pinned for a reason. Reason being mostly lack of anyone willing to update HAL for latest framework, test it and submit PR, but still.

mtk11, 2022-12-20:
Under VSC it does not compile too.

```
Windows PowerShell
Copyright (C) 2009 Microsoft Corporation. All rights reserved.

PS C:\Marlin-bugfix-2.1.x> platformio run --silent -e FYSETC_E4
Marlin\src\inc\Warnings.cpp:67:4: warning: #warning "Your Configuration provides no method to acquire user feedback!" [-Wcpp]
   #warning "Your Configuration provides no method to acquire user feedback!"
    ^
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
Marlin\src\module\temperature.cpp:2795:34: warning: unknown option after '#pragma GCC diagnostic' kind [-Wpragmas]
   #pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
                                  ^
Marlin\src\module\temperature.cpp: In static member function 'static void Temperature::isr()':
Marlin\src\module\temperature.cpp:3667:36: warning: unknown option aft
 diagnostic' kind [-Wpragmas]
     #pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
                                    ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
PS C:\Marlin-bugfix-2.1.x> echo "done" >"C:\Users\User\AppData
"
PS C:\Marlin-bugfix-2.1.x>
```

cbagwell, 2022-12-20:
The messages shown are warnings only and not errors/compile failures.  It looks like it finished compiling but if you take off the --silent option then success will be more obvious.

mtk11, 2022-12-20:
Thanks. It did build something.