MarlinFirmware/Marlin #25125: [BUG] MKS Gen 1.4, Switching Extruders from 1 to 2 changes FAN mosfet instead

Charmandrigo, 2022-12-20:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Upon remaking my code to upgrade to 2.1.2 on my modded Anet A8 Plus with a MKS Gen 1.4, I noticed my fans stopped working when I set the speed to 100%
upon a lot of reviewing on my code I realized changing the Extruders from 2 back to 1 makes the fan speed work again, switching to EXTRUDERS 2 makes the fan again stop working but instead, changing E1 Temperature boots up the fans instead of the E1 Mosfet, while the extruder remains cold meaning the E1 mosfet is not turning on at all and the function was switched to the FAN mosfet. Not even changing fan speed to 100% did anything to the second extruder (assuming the E1 and FAN mosfets switched places)
This is the schematic of my MKS Gen 1.4 with two extruders
[image]

Note that I did not modified pins_ramps at all, and I only added Anet Display support to pins_MKS_GEN_13, the issue persist even if I use unmodified pins_MKS_GEN_13.

### Bug Timeline

When upgrading from 2.0.9.5 to 2.1.2

### Expected behavior

fans to work and E1 to heat on the assigned E1 mosfet of the motherboard

### Actual behavior

FAN mosfet does the E1 heater function instead, while E1 mosfet does nothing at all

### Steps to Reproduce

1. Upload Marlin 2.1.2 configuration setup with two extruders

### Version of Marlin Firmware

2.1.2

### Printer model

Anet A8 Plus

### Electronics

MKS Gen v1.4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I attach my configurations and pins files used
[config 2.1.2.zip](https://github.com/MarlinFirmware/Marlin/files/10272577/config.2.1.2.zip)

Note that by simply switching `#define  EXTRUDERS 2` back to 1, the fan will do it's expected behaviour

ellensp, 2022-12-20:
you have #define EXTRUDERS 1

Charmandrigo, 2022-12-20:
> you have #define EXTRUDERS 1

Yes I mean that's the setting I have, just swith that to 2 and the fans will stop working
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/10273035/Configuration.h.zip)

ellensp, 2022-12-20:
Some pin debugging
```
EXTRUDERS 1

PIN:   7   Port: H4        MOSFET_D_PIN                           Input  = 0    TIMER4B   PWM:     0    WGM: 14    COM4B: 2    CS: 1    TCCR4A: 2    TCCR4B: 25    TIMSK4: 0  
PIN:   8   Port: H5        HEATER_BED_PIN                         protected 
.                          MOSFET_C_PIN                           protected 
PIN:   9   Port: H6        FAN_PIN                                protected 
.                          MOSFET_B_PIN                           protected 
PIN:  10   Port: B4        HEATER_0_PIN                           protected 
.                          MOSFET_A_PIN                           protected 

```
```
EXTRUDERS 2

PIN:   7   Port: H4        MOSFET_D_PIN                           Input  = 0    TIMER4B   PWM:     0    WGM: 14    COM4B: 2    CS: 1    TCCR4A: 2    TCCR4B: 25    TIMSK4: 0  
PIN:   8   Port: H5        HEATER_BED_PIN                         protected 
.                          MOSFET_C_PIN                           protected 
PIN:   9   Port: H6        HEATER_1_PIN                           protected 
.                          MOSFET_B_PIN                           protected 
PIN:  10   Port: B4        HEATER_0_PIN                           protected 
.                          MOSFET_A_PIN                           protected 

```
FAN_PIN is not defined with 2 Extruders

ellensp, 2022-12-20:
In Marlin/src/pins/ramps/pins_MKS_GEN_13.h

change

// Power outputs EFBF or EFBE
#define MOSFET_D_PIN                           7

to

// Power outputs EFBF or EFBE
#define MOSFET_B_PIN                           7
#define FAN_PIN                                9

This results in the following pins

```
EXTRUDERS 1
PIN:   7   Port: H4        MOSFET_B_PIN                           Input  = 0    TIMER4B   PWM:     0    WGM: 14    COM4B: 2    CS: 1    TCCR4A: 2    TCCR4B: 25    TIMSK4: 0  
PIN:   8   Port: H5        HEATER_BED_PIN                         protected 
.                          MOSFET_C_PIN                           protected 
PIN:   9   Port: H6        FAN_PIN                                protected 
PIN:  10   Port: B4        HEATER_0_PIN                           protected 
.                          MOSFET_A_PIN                           protected 

```
```
EXTRUDERS 2
PIN:   7   Port: H4        HEATER_1_PIN                           protected 
.                          MOSFET_B_PIN                           protected 
PIN:   8   Port: H5        HEATER_BED_PIN                         protected 
.                          MOSFET_C_PIN                           protected 
PIN:   9   Port: H6        FAN_PIN                                protected 
PIN:  10   Port: B4        HEATER_0_PIN                           protected 
.                          MOSFET_A_PIN                           protected 

```

ellensp, 2022-12-20:
A PR has been created to fix this.

closed 2022-12-20 by ellensp

Charmandrigo, 2022-12-20:
Thanks! The changes worked! I wonder why this wasn't an issue on 2.0.9