MarlinFirmware/Marlin #25129: Bad behaviour when EVENT_GCODE_SD_ABORT "G27 P2" is combined with Input Shaper

Mjankor, 2022-12-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have 
#define EVENT_GCODE_SD_ABORT "G27 P2" 
Park position is 0,0 (approximately X_min position and Y_min position)

When using Input shaper - upon abort the printhead runs into the frame.

When input shaper is disabled - upon abort the printer parks its head properly.

### Bug Timeline

2.1.2 (and bugfix 21/12/22)

### Expected behavior

I expected the printhead to park using "G27 P2" when print is aborted and Input shaper is enabled.

### Actual behavior

Printhead crashes into printer frame if print is aborted and input shaper is enabled.. 

### Version of Marlin Firmware

2.1.2 and 2.1.bugfix

### Printer model

Custom CoreXY

### Electronics

SKR 1.4, 2209s, PINDA, 

### Add-ons

NA

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config files.zip](https://github.com/MarlinFirmware/Marlin/files/10277081/Config.files.zip)

cbagwell, 2022-12-21:
I was not able to reproduce this myself.  I set #define EVENT_GCODE_SD_ABORT "G27 P2" and aborted an SD print and I also added it in Octoprint config and aborted a host print.  I tested 2 aborted prints each way and all parked as expected with input shaping enabled.

I'm slowly debugging the occasional failed homing (crashes into end stop) since I upgraded to a SKR Mini E3 V3 and homing after an abort (without parking first) is the most common time it fails for me.  I thought this SD abort might be a related way to reproduce the issue.  I also use ABL instead of UBL.

Mjankor, 2022-12-22:
Hmm. That's annoying. I was afraid it was going to be a problem that isn't easy to replicate. I'll swap the G27 P2 out for G28XY and see if that behaves.

Mjankor, 2022-12-29:
Just a couple more updates.
Reinstalled 2.1.1 then 2.1.2 - no change.
Used G27 instead of G27 P2 - slight change. There's less movement in the X-axis, with most of the movement in the Y-axis before crashing into frame.
Used G28XY - seems to work. Will double check later.

Mjankor, 2023-01-02:
Stopped a print today and before it managed to home I saw it's X/Y coordinates. X-axis made sense, but Y-axis reported ~2400mm. I suspect this is why parking has been borked - using a dodgy Y-axis coordinate when aborting print.

So what's happening at the end of the print that can give an incorrect Y-axis coordinate.
Possible things uncommon to my build.
Limit switch (sensorless) is set to Y-axis max , not Y-min.
Abort on endstops hit is on in firmware, but disabled.
Input shaping is on (and the problem seems to go away when input shaping is disabled).

Mjankor, 2023-01-14:
I spotted and was able to replicate the cause of the weird behaviour. Here's a video. As you can see, the X and Y position is updated immediately after the abort to incorrect coordinates (From 186, 152 (sensible) to 316,542 (wrong)). 

https://user-images.githubusercontent.com/17077516/212459515-9dc7b34c-0114-41e3-9f96-6a3ceb3079ab.mov

cbagwell, 2023-01-15:
Sure does look like a bug.  Was that Z raise to 4 a sensible value (part of parking)? 

Thats my guess on area the bug could be.  I use Bilinear leveling (and don't see this issue) and since that code is so different then UBL perhaps it's something about the UBL version of Z fade thats corrupting the X/Y values.  Does the abort issue only become visible if you abort at layers below your fade height (10mm)?

thisiskeithb, 2023-01-15:
Input Shaping support for CoreXY will be blocked in the next release and is now blocked in `bugfix-2.1.x`, so disable Input Shaping and see if the issue persists.

Mjankor, 2023-01-15:
I understand the rationale of why input shaping is blocked for CoreXY (CoreXY should be done per-axis, not per-stepper). However, I don't see why that rationale would have any bearing on this bug. Is there some other issue with CoreXY kinematics and stepper based IS that I'm missing?

Mjankor, 2023-01-15:
> Sure does look like a bug. Was that Z raise to 4 a sensible value (part of parking)?
>
> Thats my guess on area the bug could be. I use Bilinear leveling (and don't see this issue) and since that code [...]

Not sure, but I wouldn't be surprised if that Z-value is wrong too. 
I doubt it's UBL related though. I've been running that for several releases. I think it's IS related and my hunch is it's related to the ShapingQueue:purge function. I'm still trying to figure out what Marlin does when you call "Cancel Print" from the menu.

Mjankor, 2023-02-11:
Slowly getting closer. M410 has the same bug, so it's something in quickstop_stepper();

Quickstop_stepper in motion.cpp is
`void quickstop_stepper() {
  planner.quick_stop();
  planner.synchronize();
  set_current_from_steppers_for_axis(ALL_AXES_ENUM);
  sync_plan_position();
}`

planner.synchronise, set_current_from_steppers_for_axis, and sync_plan_position are called all the time, so they are not likely to be at fault. I suspect it's in planner.quick_stop(). But, that code in planner.cpp hasn't been changed over the last couple of marlin versions. It does call stepper.quick_stop() in stepper.cpp which has been updated. stepper.quick_stop sets the abort_current_block flag, which enables this code.

  // If we must abort the current block, do so!
  if (abort_current_block) {
    abort_current_block = false;
    if (current_block) {
      discard_current_block();
      #if HAS_SHAPING
        ShapingQueue::purge();
        #if ENABLED(INPUT_SHAPING_X)
          shaping_x.delta_error = 0;
          shaping_x.last_block_end_pos = count_position.x;
        #endif
        #if ENABLED(INPUT_SHAPING_Y)
          shaping_y.delta_error = 0;
          shaping_y.last_block_end_pos = count_position.y;
        #endif
      #endif
    }
  }
  
  Now to dig into the shaping code and see what shaping_x.last_block_end_pos = count_position.x; does
  
  Please excuse my hackiness. I'm not a coder. :D

cbagwell, 2023-02-13:
I think now that you're able to narrow it down that it is pointing more towards CoreXY logic and that Input Shaping purge logic misbehaving together.

Mjankor, 2023-02-13:
Ahh. X, Y and A, B axes/motors mismatch?

cbagwell, 2023-02-13:
Thats what I'm thinking since that seems the be the main item that stands out about your config.

I'm not familiar with CoreXY logic in Marlin but in the code block you quoted it references current_position.x/y and last_block_end_pos similar to what's being done in _set_position().  I notice in _set_position() there is some logic related to CoreXY messing with A/B values and then updates last_block_end_pos.  Perhaps this abort path needs similar logic to translate A/B and then update last_block_end_pos.

closed 2023-04-25 by github-actions[bot]