MarlinFirmware/Marlin #25132: [BUG] Trouble build, stops at moduel temperareratur in diffrent ways

3DSwed, 2022-12-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Get error message as follow, dont now whats wrong. At first got the same type of error on pid for hotend. Turn that off to try and get a working build. Now the folling error comes up and i have also turned bed pid off but no changes.
Been scratching my head for a couple of days now... last i looked at was if the chosing of tempsensor had any to do with it. No change. Turn Bed pid off as well know next issue with Ejerk in same module... 
I dont know if it´s me or a bug, but saw another thread that got issue withe the same module.

Compiling .pio\build\LPC1768\src\src\module\temperature.cpp.o
In file included from Marlin\src\module\../inc/MarlinConfig.h:47,
                 from Marlin\src\module\settings.h:28,
                 from Marlin\src\module\settings.cpp:46:
Marlin\src\module\settings.cpp: In static member function 'static bool MarlinSettings::save()':
Marlin\src\module\settings.cpp:795:70: error: 'DEFAULT_EJERK' was not declared in this scope
  795 |         const xyze_pos_t planner_max_jerk = LOGICAL_AXIS_ARRAY(float(DEFAULT_EJERK), 10, 10, 0.4, 0.4, 0.4, 0.4, 0.4, 0.4, 0.4);
      |                                                                      ^~~~~~~~~~~~~
Marlin\src\module\../inc/../core/types.h:78:28: note: in definition of macro 'LIST_ITEM_E'
   78 |   #define LIST_ITEM_E(N) , N
      |                            ^
Marlin\src\module\../inc/../core/types.h:58:38: note: in expansion of macro 'LOGICAL_AXIS_LIST'
   58 | #define LOGICAL_AXIS_ARRAY(E,V...) { LOGICAL_AXIS_LIST(E,V) }
      |                                      ^~~~~~~~~~~~~~~~~
Marlin\src\module\settings.cpp:795:45: note: in expansion of macro 'LOGICAL_AXIS_ARRAY'
  795 |         const xyze_pos_t planner_max_jerk = LOGICAL_AXIS_ARRAY(float(DEFAULT_EJERK), 10, 10, 0.4, 0.4, 0.4, 0.4, 0.4, 0.4, 0.4);
      |                                             ^~~~~~~~~~~~~~~~~~
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\settings.h:28,
                 from Marlin\src\module\settings.cpp:46:
Marlin\src\module\settings.cpp: In static member function 'static void MarlinSettings::reset()':
Marlin\src\module\settings.cpp:3131:29: error: 'DEFAULT_Kp' was not declared in this scope
 3131 |           ARRAY_BY_HOTENDS1(DEFAULT_Kp)
      |                             ^~~~~~~~~~
Marlin\src\module\../inc/../core/macros.h:366:24: note: in definition of macro 'LIST_1'
  366 | #define LIST_1( A,...) A
      |                        ^
Marlin\src\module\../inc/../core/macros.h:371:23: note: in expansion of macro '_LIST_N'
  371 | #define LIST_N_1(N,K) _LIST_N(N,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K)
      |                       ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:373:27: note: in expansion of macro 'LIST_N_1'
  373 | #define ARRAY_N_1(N,K)  { LIST_N_1(N,K) }
      |                           ^~~~~~~~
Marlin\src\module\../inc/Conditionals_LCD.h:1032:31: note: in expansion of macro 'ARRAY_N_1'
 1032 | #define ARRAY_BY_HOTENDS1(v1) ARRAY_N_1(HOTENDS, v1)
      |                               ^~~~~~~~~
Marlin\src\module\settings.cpp:3131:11: note: in expansion of macro 'ARRAY_BY_HOTENDS1'
 3131 |           ARRAY_BY_HOTENDS1(DEFAULT_Kp)
      |           ^~~~~~~~~~~~~~~~~
Marlin\src\module\settings.cpp:3137:29: error: 'DEFAULT_Ki' was not declared in this scope
 3137 |           ARRAY_BY_HOTENDS1(DEFAULT_Ki)
      |                             ^~~~~~~~~~
Marlin\src\module\../inc/../core/macros.h:366:24: note: in definition of macro 'LIST_1'
  366 | #define LIST_1( A,...) A
      |                        ^
Marlin\src\module\../inc/../core/macros.h:371:23: note: in expansion of macro '_LIST_N'
  371 | #define LIST_N_1(N,K) _LIST_N(N,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K)
      |                       ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:373:27: note: in expansion of macro 'LIST_N_1'
  373 | #define ARRAY_N_1(N,K)  { LIST_N_1(N,K) }
      |                           ^~~~~~~~
Marlin\src\module\../inc/Conditionals_LCD.h:1032:31: note: in expansion of macro 'ARRAY_N_1'
 1032 | #define ARRAY_BY_HOTENDS1(v1) ARRAY_N_1(HOTENDS, v1)
      |                               ^~~~~~~~~
Marlin\src\module\settings.cpp:3137:11: note: in expansion of macro 'ARRAY_BY_HOTENDS1'
 3137 |           ARRAY_BY_HOTENDS1(DEFAULT_Ki)
      |           ^~~~~~~~~~~~~~~~~
Marlin\src\module\settings.cpp:3143:29: error: 'DEFAULT_Kd' was not declared in this scope
 3143 |           ARRAY_BY_HOTENDS1(DEFAULT_Kd)
      |                             ^~~~~~~~~~
Marlin\src\module\../inc/../core/macros.h:366:24: note: in definition of macro 'LIST_1'
      |                        ^
Marlin\src\module\../inc/../core/macros.h:371:23: note: in expansion of macro '_LIST_N'
  371 | #define LIST_N_1(N,K) _LIST_N(N,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K,K)
      |                       ^~~~~~~
Marlin\src\module\../inc/../core/macros.h:373:27: note: in expansion of macro 'LIST_N_1'
  373 | #define ARRAY_N_1(N,K)  { LIST_N_1(N,K) }
      |                           ^~~~~~~~
Marlin\src\module\../inc/Conditionals_LCD.h:1032:31: note: in expansion of macro 'ARRAY_N_1'
 1032 | #define ARRAY_BY_HOTENDS1(v1) ARRAY_N_1(HOTENDS, v1)
      |                               ^~~~~~~~~
Marlin\src\module\settings.cpp:3143:11: note: in expansion of macro 'ARRAY_BY_HOTENDS1'
 3143 |           ARRAY_BY_HOTENDS1(DEFAULT_Kd)
      |           ^~~~~~~~~~~~~~~~~
*** [.pio\build\LPC1768\src\src\module\settings.cpp.o] Error 1

### Bug Timeline

3 Days 

### Version of Marlin Firmware

2.1.x Bugfix (latest (2.1.2))

### Printer model

Home made 

### Electronics

Re_Arm LPC1768

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/10278792/Configuration_adv.zip)

thisiskeithb, 2022-12-21:
> error: 'DEFAULT_EJERK' was not declared in this scope

Problem: You disabled `DEFAULT_EJERK` which causes this error.

Solution: Enable `DEFAULT_EJERK`.

> error: 'DEFAULT_Kp' was not declared in this scope

Problem: You enabled `PID_PARAMS_PER_HOTEND`, but disabled the `DEFAULT_K*_LIST` values (and you only have a single hotend, so this define/feature doesn't apply).

Solution: Disable `PID_PARAMS_PER_HOTEND`:

<details><summary>[SUCCESS] Took 41.49 seconds:</summary>

```prolog
Building .pio/build/LPC1768/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  42.7% (used 13980 bytes from 32736 bytes)
Flash: [====      ]  41.8% (used 198412 bytes from 475136 bytes)
=============================== [SUCCESS] Took 41.49 seconds ===============================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        SUCCESS   00:00:41.485
=============================== 1 succeeded in 00:00:41.485 ===============================
 *  Terminal will be reused by tasks, press any key to close it.
```
</details>

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-12-21 by thisiskeithb

3DSwed, 2022-12-21:
Thank you so mutch that solved my misstakes. Now i can take the next step and try the  2208 Smart shapers i got. I did post on two of the above sugested as well but no answers...