MarlinFirmware/Marlin #25133: SKR v3 won't compile on 2.1.2 with new platform environment

gordo3di, 2022-12-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Using the new ststm32 15.4.1 platform the compiler just hangs. I had to downgrade to the older 14.1.0 platform to get it to work. I thought it was something with platformio so i uninstalled everything and started from scratch and still had the issue.

In stm32h7.ini

```c
#
# BigTreeTech SKR V3.0 / V3.0 EZ (STM32H743VIT6 ARM Cortex-M7)
#
[env:STM32H743Vx_btt]
extends                     = stm32_variant
#platform                    = ststm32@~15.4.1
platform                    = ststm32@~14.1.0
#platform_packages           = framework-arduinoststm32@~4.20200.220530
platform_packages           = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32/archive/main.zip
board                       = marlin_STM32H743Vx
```

### Bug Timeline

Marlin 2.1.2

### Expected behavior

Should compile

### Actual behavior

Just hangs forever

### Steps to Reproduce

Select the new skr v3 board in configuration.h and platformio.ini

### Version of Marlin Firmware

2.1.2

### Printer model

gMax 2 PRO

### Electronics

Skr v3 

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[221221_config.zip](https://github.com/MarlinFirmware/Marlin/files/10279862/221221_config.zip)

thisiskeithb, 2022-12-21:
This config fails to build with all kinds of issues starting with:

<details><summary>No E stepper plug left for Y2/Z2:</summary>

```prolog
Executing task: platformio run 

Processing STM32H743Vx_btt (platform: ststm32@~15.4.1; board: marlin_STM32H743Vx; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:926,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins_postprocess.h:863:8: error: #error "No E stepper plug left for Y2!"
  863 |       #error "No E stepper plug left for Y2!"
      |        ^~~~~
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins_postprocess.h:949:8: error: #error "No E stepper plug left for Z2!"
  949 |       #error "No E stepper plug left for Z2!"
      |        ^~~~~
Error: Failed to parse Marlin features. See previous error messages.
```
</details>

The SKR 3 board only has five driver slots and you've enabled seven.

Commenting out the extra enabled drivers lets the build move on but fails a short time later:

<details><summary>BLTOUCH_DELAY/GRID_MAX_POINTS_[XY]/Thermistor 66/BLTOUCH_DELAY errors/etc:</summary>

```prolog
In file included from Marlin/src/HAL/STM32/../../inc/MarlinConfig.h:49,
                 from Marlin/src/HAL/STM32/HAL.cpp:27:
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1778:8: error: #error "BLTOUCH_DELAY less than 200 is unsafe and is not supported."
 1778 |       #error "BLTOUCH_DELAY less than 200 is unsafe and is not supported."
      |        ^~~~~
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:1993:6: error: #error "GRID_MAX_POINTS_[XY] must be a whole number between 3 and 15."
 1993 |     #error "GRID_MAX_POINTS_[XY] must be a whole number between 3 and 15."
      |      ^~~~~
Marlin/src/HAL/STM32/../../inc/SanityCheck.h:2351:6: error: #error "Thermistor 66 requires HEATER_0_MINTEMP > 20."
 2351 |     #error "Thermistor 66 requires HEATER_0_MINTEMP > 20."
      |      ^~~~~
```
</details>

Using a known-good SKR 3 build, it compiles fine.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
- [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
- [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group 
- [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2022-12-21 by thisiskeithb

gordo3di, 2022-12-21:
> This config fails to build with all kinds of issues starting with:
> 
> No E stepper plug left for Y2/Z2:
> The SKR 3 board only has five driver slots and you've enabled seven.
> 
> Commenting out the extra enabled drivers lets the build move on but fails a short time later:
> 
> BLTOUCH_DELAY/GRID_MAX_POINTS_[XY]/Thermistor 66/BLTOUCH_DELAY errors/etc:
> Using a known-good SKR 3 build, it compiles fine.
> 
> This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:
> 
> * [Marlin Documentation](https://marlinfw.org) hosted on MarlinFW.org
> * [Marlin Forum](https://forums.reprap.org/list.php?415) hosted on RepRap.org
> * [Marlin Firmware](https://www.facebook.com/groups/1049718498464482/) Facebook Group
> * [Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
> * [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
> * [Marlin Firmware](https://discord.gg/n5NJ59y) Discord server
> 
> After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

I'm using the BTT expansion board set in the pins file since I have separate Y and Z motors, I removed the bltouch min time sanity check since I set it to 10 for my builds (because 200 isn't necessary) and ignored the 66 thermistor since it has all sorts of issues from the previous version (on the skr v3 which I've spend over 3 days working on the thermistor tables).

I have 45 skr v3's and I put it on one that works fine with marlin 2.1.1. It also only compiles when I reduce the version. 
Is there something else that might cause this to fail?

thisiskeithb, 2022-12-21:
> I'm using the BTT expansion board set in the pins file since I have separate Y and Z motors,

`BTT_MOTOR_EXPANSION` wasn't enabled in your config, so that's why the build errors out.

> I removed the bltouch min time sanity check since I set it to 10 for my builds (because 200 isn't necessary) 

Don't do this. We add these sanity checks for a reason.

> ignored the 66 thermistor since it has all sorts of issues from the previous version

I'm not sure I follow, but if you want to open a separate issue (or PR with a fix), that'd be great.

> Is there something else that might cause this to fail?

Delete the hidden `.pio` folder at the root of your Marlin project and try compiling again.

If that doesn't work:

1. Delete the hidden `.platformio` folder from your User directory:
 
    - Windows: `%USERPROFILE%\.platformio\`
    - macOS, Unix, Unix-like: `~/.platformio/`

2. Re-install PlatformIO.

This isn't a support forum, so please use the links in my previous post for further help.

gordo3di, 2022-12-22:
You got it Keith. I will never post a possible bug on here ever again and I definitely will never post my configs. The build didn't error out, it literally hanged for 25 minutes unless I changed the version but whatever.