MarlinFirmware/Marlin #25134: [BUG] Hephestos 2: Z axis not working after commit 89334caa526f2d300eee834d34d06d8f837a57d5

Bergerac56, 2022-12-21:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

With the last bugfix, my Hephestos 2 does not work anymore.	
Problem with Z axis. X & Y seem to work normally (as far as it can be tested without homing). Z axis does not work at all anymore (shaking but not moving)

**Retroacts:** 
- For personal reasons I could not follow new developments since 2022-09-04 bump distribution date. At that time, all was working fine. 
- First attempt to upgrade at bump distribution date 2022-12-14: The printer did not work anymore.

**What I did:** 

- I checked carefully the modifications of my config files and compared with the proposed Hephestos 2 files (@jbrazio): no result. 
- I took a fresh copy of marlin and just replaced the config files with those proposed for the Hephestos 2, (same release, from examples configs and maintained by @jbrazio): The printer does not work too (in this case, using only “marlin stuffs”).
- I did the same test with the freshest release of Bugfix (today), same result.

**Trying to identify the cause:**

- Starting from the last “bump distribution date” known as working for me (2022-09-14), I checked commit per commit and found commit “Input Shaping Improvements” #24951 (2022-11-29) (commit 89334caa526f2d300eee834d34d06d8f837a57d5) to be the problem. 
- Without this commit (reversed) and using the last bugfix release of today, the Z axis is again functional.  BUT the overall movements of X and Y axis are very trash and the printer vibrates a lot. (Due apparently to the fact that Input Shaping Improvements had a lot to do with vibrations).

**Conclusion:** 

- Did I forget something? Is this a known problem? (Did not find anything at first search) Or is it just me?

### Bug Timeline

occured after commit 89334caa526f2d300eee834d34d06d8f837a57d5

### Expected behavior

Z axis should work normally

### Actual behavior

Z axis stepper motors are shaking (vibrating) but not moving

### Steps to Reproduce

it still occures with the last bugfix version and the example config files provided for  BQ Hephestos 2.  And, obviously, with my config files.

### Version of Marlin Firmware

Bugfix 2.1.x  2022-12-21

### Printer model

BQ HEPHESTOS 2

### Electronics

BQ ZUM

### Add-ons

BLTOUCH

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Cavalletta98, 2022-12-23:
I got the same problem with the release 2.1.2. Have you solved it?

Bergerac56, 2022-12-23:
Unfortunately not yet. I am still digging in all the modifications related to shaping. But still no solution. The last version OK for me is the bugfix 2.1 bump date 2022-11-28.

Cavalletta98, 2022-12-27:
@Bergerac56  Are you on discord? because i'm getting some problem with marlin version 2.1.1. In particular i have some problems during printing. If you changed something from the original configuration file, can you please share it with me?

Cavalletta98, 2022-12-28:
Can i use your config file with fw 2.1.1?

Bergerac56, 2022-12-28:
@Cavalletta98 Hello. Yes you can try.  

I join  a set of files with less "personnal comments" inside ;)

Those files are adapted for my own use and are only tested with a BQ Hephestos 2 modified for a BLTOUCH (The original Z sensor "died" very quickly and BQ was really not helpfull.)

The set of files contains 2 files with the correct pinout for the BLTOUCH (place them in the "pins" directory) and the config files wich are working nicely for me. The files were adapted for the last bugfix still functionning for me (2022-11-28) and  use my own parameters for the position of the Z sensor. If it works, you will have to recalibrate Z offset for your use. 

It is obvious, but do not forget to reload the factory defaults after the upload of the adapted firmware. And watch the printer during the first use. Who knows what can happen :)
Good luck

[Bergerac56 Config files.zip](https://github.com/MarlinFirmware/Marlin/files/10315548/Bergerac56.Config.files.zip)

Cavalletta98, 2023-01-01:
@Bergerac56 As a curiosity, why do you change `DEFAULT_AXIS_STEPS_PER_UNIT` from standard values?

Bergerac56, 2023-01-02:
after having calculated the exact moves and extrusion on my printer, it was more appropriate.

Cavalletta98, 2023-01-02:
@Bergerac56 Which value do you use for offset Z probe?

Bergerac56, 2023-03-04:
It seems that the issue #25117 concerns the same problem. I will close this issue

closed 2023-03-04 by Bergerac56