MarlinFirmware/Marlin #25135: [BUG] Disabling elapsed time still shows

RustedAperture, 2022-12-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

When disabling elapsed time by commenting it out, it still shows on the ender 3 v2 display. Have not tested with other displays.

I have enabled only remaining time but elapsed still shows.

### Bug Timeline

not sure

### Expected behavior

only remaining time to show

### Actual behavior

cycles between remaining and elapsed

### Steps to Reproduce

1. comment out `#define SHOW_ELAPSED_TIME`
2. remove comment from `#define SHOW_REMAINING_TIME`

### Version of Marlin Firmware

2022-12-21

### Printer model

Ender 3 V2

### Electronics

SKR Mini E3 V2, Stock Ender 3 screen

### Add-ons

BLTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10283896/Marlin.zip)

RustedAperture, 2022-12-22:
Initialized the EEPROM and the problem was solved, however, now the LCD is not usable during the print.

ellensp, 2023-02-21:
running DWIN_MARLINUI_PORTRAIT on Stock Ender 3 **V2** screen

ellensp, 2023-02-21:
looking at https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/lcd/e3v2/marlinui/ui_status_480x272.cpp#L369-L422

From looking at the code this is simply not been implemented yet

RustedAperture, 2023-02-21:
> running DWIN_MARLINUI_PORTRAIT on Stock Ender 3 **V2** screen

Yes, sorry should have specified since there are multiple available for the DWIN E3V2 screen.

RustedAperture, 2023-02-21:
> looking at https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.1.x/Marlin/src/lcd/e3v2/marlinui/ui_status_480x272.cpp#L369-L422
> 
> From looking at the code this is simply not been implemented yet

Is it ok to change the title to [FR] then?

ellensp, 2023-02-21:
Yes I think that would be appropriate.