MarlinFirmware/Marlin #25136: Support for Blackpill (STM32F401CDU6)

arslan437, 2022-12-22:
### Is your feature request related to a problem? Please describe.

I am trying to make a custom control board based on black pill(STM32F401CDU6) because it widely available and cheep. 

 

### Are you looking for hardware support?

I have checked the marlin firmware, it do have creality  board based on STM32F401RC. Both of these controllers have some pin difference. 

Is there any way I can compile the code for black pill ? 

### Describe the feature you want

I am using platformio to compile the code. 

Here is the same ini file which the black pill works fine. 

```
[env:blackpill_f401cc]
platform = ststm32
board = blackpill_f401cc
framework = arduino

upload_protocol = stlink

```

### Additional context

I am using a high precision stepper driver which have around 10,000 step per rotation. That's why I want to use stm32.

Bob-the-Kuhn, 2022-12-22:
Be nice to yourself and get a BTT or MKS STM32F4xx based controller.  Much less hassle and not much added expense.

arslan437, 2022-12-22:
I want to make SLA printer not FDM. That's why I want to make a custom board. I will be using the nano DLP for that.

EvilGremlin, 2022-12-22:
This is how you add board: https://github.com/MarlinFirmware/Marlin/pull/24791/files
or you can set all defines in env with -Dflags but ew

arslan437, 2022-12-22:
I have checked the link that you share. 

This do provide lots very helpful information. I will fork the repository and try to add new board. I haven't done that before. It would great if you can guide me if stuck at some where. 

Currently I have no idea what the ldscript.ld id for.

EvilGremlin, 2022-12-22:
It's linker script, you don't have to touch it.
Basically SKIPR variant is just an edited copy of generic MARLIN_F407VE variant (or MARLIN_F4x7Vx, don't remember which), you leave only stuff you actually use enabled to eleminate ambuguities.

arslan437, 2022-12-23:
@EvilGremlin  One more question. do I have to update any thing in the variant folder?

EvilGremlin, 2022-12-23:
Nope, just add.

arslan437, 2022-12-23:
Should I add theses line in the file or not 

```
board_build.offset          = 0x10000
board_upload.offset_address = 0x08010000
```

EvilGremlin, 2022-12-23:
yep, in ini or board json

arslan437, 2022-12-23:
I think it has 96K ram

I have add all things and when I compiled I got this error

```
In file included from d:\pst\opensource\marlin\marlin\src\inc\marlinconfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:29:
d:\pst\opensource\marlin\marlin\src\pins\pins.h:890:6: error: #error "Unknown MOTHERBOARD value set in Configuration.h."
  890 |     #error "Unknown MOTHERBOARD value set in Configuration.h."
      |      ^~~~~
Error: Failed to parse Marlin features. See previous error messages.
```

here is the line from the pin.h 

```
#elif MB(TRONXY_V10)
  #include "stm32f4/pins_TRONXY_V10.h"          // STM32F4                                env:STM32F446_tronxy
#elif MB(BOARD_CUSTOM_BLACKPILL)
  #include "stm32f4/pins_custom_blackpill.h"    // STM32F4                                

```

here is the code from the board.h 

```
#define BOARD_TRONXY_V10              4244  // TRONXY V10 (STM32F446ZE)
#define BOARD_CUSTOM_BLACKPILL        4245  // Custom board based on STM32F401CDU6.
```

here is the code from the ini file. I did copied all this from creality stm32f401rc.

```
#
# Blackpill
#
[env:STM32F401CD_blackpill_stlink]
extends                     = stm32_variant
board                       = marlin_custom_blackpill
board_build.variant         = MARLIN_CUSTOM_BLACKPILL
board_build.offset          = 0x10000
board_upload.offset_address = 0x08010000
build_flags                 = ${stm32_variant.build_flags} -DMCU_STM32F401CD -DSTM32F4
                              -DSS_TIMER=4 -DTIMER_SERVO=TIM5
                              -DENABLE_HWSERIAL3 -DTRANSFER_CLOCK_DIV=8
build_unflags               = ${stm32_variant.build_unflags} -DUSBCON -DUSBD_USE_CDC
extra_scripts               = ${stm32_variant.extra_scripts}
                              pre:buildroot/share/PlatformIO/scripts/random-bin.py
monitor_speed               = 115200

debug_tool                  = stlink
upload_protocol             = stlink
```

ellensp, 2022-12-23:
not #elif MB(BOARD_CUSTOM_BLACKPILL)
you need #elif MB(CUSTOM_BLACKPILL)

EvilGremlin, 2022-12-23:
and no env in pins.h
you don't use native USB?

arslan437, 2022-12-23:
@ellensp  it fixed that issue. 

I don't get it "and no env in pins.h"

not i am using external usb to uart converter 

here is the next error

```
Error: Build environment 'STM32F401CD_blackpill_stlink' is incompatible with BOARD_CUSTOM_BLACKPILL. Use one of these: 
```

EvilGremlin, 2022-12-23:
look at comment in pins.h

ellensp, 2022-12-23:
see example 
  #include "stm32f4/pins_TRONXY_V10.h"          // STM32F4                                env:STM32F446_tronxy

the comment is the list of valid env's for your board

arslan437, 2022-12-23:
sanity check is fixed 

```

Processing STM32F401CD_blackpill_stlink (board: marlin_custom_blackpill; platform: ststm32@~12.1; framework: arduino)
------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/marlin_custom_blackpill.html
PLATFORM: ST STM32 (12.1.1) > STM32F401CD (96k RAM. 384k Flash)
HARDWARE: STM32F401CDU6 84MHz, 96KB RAM, 384KB Flash
DEBUG: Current (stlink) External (jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 4.10900.200819 (1.9.0)
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IWatchdog @ 1.0.0   
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1      
|-- SoftwareSerial @ 1.0
|-- Wire @ 1.0
|-- Servo @ 1.1.2
Building in release mode
Compiling .pio\build\STM32F401CD_blackpill_stlink\FrameworkArduinoVariant\PeripheralPins.c.o
Compiling .pio\build\STM32F401CD_blackpill_stlink\FrameworkArduinoVariant\variant.cpp.o
Compiling .pio\build\STM32F401CD_blackpill_stlink\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F401CD_blackpill_stlink\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F401CD_blackpill_stlink\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F401CD_blackpill_stlink\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F401CD_blackpill_stlink\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F401CD_blackpill_stlink\FrameworkArduino\Tone.cpp.o
In file included from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/stm32_def.h:34,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/clock.h:43,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino/wiring_time.h:23,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino/wiring.h:38,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino/Arduino.h:36,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\variants\marlin_custom_blackpill\PeripheralPins.c:31:
C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f4xx.h:191:3: error: #error "Please select first the target STM32F4xx device used in your application (in stm32f4xx.h file)"
  191 |  #error "Please select first the target STM32F4xx device used in your application (in stm32f4xx.h file)"  
      |   ^~~~~
In file included from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/stm32_def.h:34,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/PortNames.h:33,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/PinNames.h:5,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino/pins_arduino.h:24,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\variants\marlin_custom_blackpill\variant.cpp:19:
C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f4xx.h:191:3: error: #error "Please select first the target STM32F4xx device used in your application (in stm32f4xx.h file)"
  191 |  #error "Please select first the target STM32F4xx device used in your application (in stm32f4xx.h file)"  
      |   ^~~~~
In file included from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal_rcc.h:33,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\variants\marlin_custom_blackpill/hal_conf_custom.h:280,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\STM32F4xx/stm32f4xx_hal_conf.h:8,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal.h:30,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f4xx.h:250,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/stm32_def.h:34,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino\stm32/clock.h:43,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino/wiring_time.h:23,
                 from C:\Users\M.Arsalan\.platformio\packages\framework-arduinoststm32@4.10900.200819\cores\arduino/wiring.h:38,
[... cut ...]
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F401CD_blackpill_stlink\FrameworkArduino\Tone.cpp.o] Error 1
========================================== [FAILED] Took 16.30 seconds ==========================================

Environment                   Status    Duration
----------------------------  --------  ------------
STM32F401CD_blackpill_stlink  FAILED    00:00:16.302
===================================== 1 failed, 0 succeeded in 00:00:16.302 ===================================== 
```

ellensp, 2022-12-23:
look in the file it has
```
** @addtogroup Device_Included
  * @{
  */

#if defined(STM32F405xx)
  #include "stm32f405xx.h"
#elif defined(STM32F415xx)
  #include "stm32f415xx.h"
#elif defined(STM32F407xx)
  #include "stm32f407xx.h"
#elif defined(STM32F417xx)
  #include "stm32f417xx.h"
#elif defined(STM32F427xx)
  #include "stm32f427xx.h"
#elif defined(STM32F437xx)
  #include "stm32f437xx.h"
#elif defined(STM32F429xx)
  #include "stm32f429xx.h"
#elif defined(STM32F439xx)
  #include "stm32f439xx.h"
#elif defined(STM32F401xC)
  #include "stm32f401xc.h"
#elif defined(STM32F401xE)
  #include "stm32f401xe.h"
#elif defined(STM32F410Tx)
  #include "stm32f410tx.h"
#elif defined(STM32F410Cx)
  #include "stm32f410cx.h"
#elif defined(STM32F410Rx)
  #include "stm32f410rx.h"
#elif defined(STM32F411xE)
  #include "stm32f411xe.h"
#elif defined(STM32F446xx)
  #include "stm32f446xx.h"
#elif defined(STM32F469xx)
  #include "stm32f469xx.h"
#elif defined(STM32F479xx)
  #include "stm32f479xx.h"
#elif defined(STM32F412Cx)
  #include "stm32f412cx.h"
#elif defined(STM32F412Zx)
  #include "stm32f412zx.h"
#elif defined(STM32F412Rx)
  #include "stm32f412rx.h"
#elif defined(STM32F412Vx)
  #include "stm32f412vx.h"
#elif defined(STM32F413xx)
  #include "stm32f413xx.h"
#elif defined(STM32F423xx)
  #include "stm32f423xx.h"
#else
 #error "Please select first the target STM32F4xx device used in your application (in stm32f4xx.h file)"
#endif

```
None are your chip, so you probably need a newer arduinoststm32

ellensp, 2022-12-23:
you should look at https://reprap.org/forum/read.php?415,889516,889539#msg-889539

arslan437, 2022-12-23:
I used the link ini configuration and some what it worked 

```
platform = ${common_stm32.platform}
extends = common_stm32
board = genericSTM32F401CD
board_build.mcu = stm32f401cdu6
board_build.f_cpu = 84000000L
upload_protocol = dfu
build_flags = ${common_stm32.build_flags} 
              -DHSE_VALUE=25000000
              -DHAL_PCD_MODULE_ENABLED
              -DHAL_UART_MODULE_ENABLED
              -DUSE_FULL_LL_DRIVER
```
```
Indexing .pio\build\STM32F401CD_blackpill_stlink\liba29\libWire.a
Archiving .pio\build\STM32F401CD_blackpill_stlink\libb96\libServo.a
Indexing .pio\build\STM32F401CD_blackpill_stlink\libb96\libServo.a
Linking .pio\build\STM32F401CD_blackpill_stlink\firmware.elf
c:/users/m.arsalan/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/STM32F401CD_blackpill_stlink/src/src/HAL/STM32/sdio.cpp.o: in function `SDIO_IRQHandler':
D:\pst\opensource\Marlin/Marlin\src\HAL\STM32/sdio.cpp:163: undefined reference to `HAL_SD_IRQHandler'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\STM32F401CD_blackpill_stlink\firmware.elf] Error 1
==================================================== [FAILED] Took 75.33 seconds ====================================================
Environment                   Status    Duration
----------------------------  --------  ------------
STM32F401CD_blackpill_stlink  FAILED    00:01:15.335
=============================================== 1 failed, 0 succeeded in 00:01:15.335 ===============================================
```

arslan437, 2022-12-23:
finally it got compiled with the following ini 

```
platform = ${common_stm32.platform}
extends = common_stm32
; board = genericSTM32F401CD
board = marlin_custom_blackpill
board_build.mcu = stm32f401cdu6
board_build.f_cpu = 84000000L
upload_protocol = dfu
build_flags = ${common_stm32.build_flags} 
              -DHSE_VALUE=25000000
              -DHAL_PCD_MODULE_ENABLED
              -DHAL_UART_MODULE_ENABLED
              -DUSE_FULL_LL_DRIVER
```

arslan437, 2022-12-23:
now issue with upload 

```
Configuring upload protocol...
AVAILABLE: dfu, jlink, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\STM32F401CD_blackpill_stlink\firmware.elf
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev-00378-ge5be992df (2020-06-26-09:29)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

embedded:startup.tcl:26: Error: Can't find scripts/interface/stlink.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
*** [upload] Error 1
============================================= [FAILED] Took 74.77 seconds =============================================
```

arslan437, 2022-12-26:
After lots of digging and testing I found that issue was in the board json file. 

For future reference 

Here is the json file with which it does not work

```
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32F401xE -DSTM32F401xx",
    "f_cpu": "84000000L",
    "hwids": [
      [
        "0x1EAF",
        "0x0003"
      ],
      [
        "0x0483",
        "0x3748"
      ]
    ],
    "ldscript": "ldscript.ld",
    "mcu": "stm32f401cdu6",
    "variant": "marlin_custom_blackpill"
  },
  "debug": {
    "jlink_device": "STM32F401CD",
    "openocd_target": "stm32f4x",
    "svd_path": "STM32F40x.svd",
    "tools": {
      "stlink": {
        "server": {
          "arguments": [
            "-f",
            "scripts/interface/stlink.cfg",
            "-c",
            "transport select hla_swd",
            "-f",
            "scripts/target/stm32f4x.cfg",
            "-c",
            "reset_config none"
          ],
          "executable": "bin/openocd",
          "package": "tool-openocd"
        }
      }
    }
  },
  "frameworks": [
    "arduino",
    "stm32cube"
  ],
  "name": "STM32F401CD (96k RAM. 384k Flash)",
  "upload": {
    "maximum_ram_size": 98304,
    "maximum_size": 393216,
    "protocol": "stlink",
    "protocols": [
      "stlink",
      "dfu",
      "jlink"
    ]
  },
  "url": "https://www.st.com/en/microcontrollers-microprocessors/stm32f401cd.html",
  "vendor": "Generic"
}
```

With which it worked 

```
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32F401xC -DSTM32F401xx",
    "f_cpu": "84000000L",
    "hwids": [
      [
        "0x1EAF",
        "0x0003"
      ],
      [
        "0x0483",
        "0x3748"
      ]
    ],
    "ldscript": "ldscript.ld",
    "mcu": "stm32f401cdu6",
    "variant": "marlin_custom_blackpill"
  },
  "debug": {
    "jlink_device": "STM32F401CD",
    "openocd_target": "stm32f4x",
    "svd_path": "STM32F40x.svd"
  },
  "frameworks": [
    "arduino",
    "stm32cube"
  ],
  "name": "STM32F401CD (96k RAM. 384k Flash)",
  "upload": {
    "maximum_ram_size": 98304,
    "maximum_size": 393216,
    "protocol": "stlink",
    "protocols": [
      "stlink",
      "dfu",
      "jlink"
    ]
  },
  "url": "https://www.st.com/en/microcontrollers-microprocessors/stm32f401cd.html",
  "vendor": "Generic"
}
```

arslan437, 2022-12-26:
Can I submit the pull request to add the board to Marlin on main repository ?

ellensp, 2022-12-26:
everything new goes into bugfix branch, from there it gets into releases

thisiskeithb, 2022-12-26:
Closing since you’ve opened a PR:

- https://github.com/MarlinFirmware/Marlin/pull/25152

closed 2022-12-26 by thisiskeithb