MarlinFirmware/Marlin #25138: [BUG] firmware crashes during G2/G3 helical interpolation on MKS TinyBee

shortedcircuit, 2022-12-22:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am using Marlin on my cnc mill. I updated the mainboard from BTT SKR 1.4 Turbo to a MKS TinyBee mainboard but the machine crashes every now and then into the workpiece.

I had these problems with 2.1.1 with bugfixes and now I am using the newest 2.1.2 firmware with bugfixes, but the problem remains.

After many tests i isolated a misbehaviour during G2/G3 helical interpolation moves. 

The task is to use an endmill with a helical tool path to create a round hole in a small series of parts.
Sometimes it works as it should, sometimes the same Gcode makes the machine to go somewhere it should not go, i.e. sudden moves to the side or down into the material.
Sometimes the ESP32 begins to lag between or during these movements, it even happened once or twice to hang itself so i had to hard reset it.

it crashes about every third part, so quite often but not in general when using G2/G3.
It had not happened (yet) with G2/G3 moves in G17 without z.

When i changed the gcode to not use helical moves but a lot of tiny G1 xyz moves, it works flawless, despite the fact that hundreds of lines of gcode are being processed per second. 

This is how a crash looks like: (I pushed the estop really quick)
[image]

When i use the same (helical movement) gcode and the same firmware on the BTT SKR 1.4 Turbo board, it won't crash either. 

This is how it should look like:
[image]

Can anyone confirm these problems with the MKS TinyBee board? 

This is what the responsible part of the gcode looks like, created from Fusion360: 

Z1.5
G1 Z0.3 F1120
G2 X2.55 Z-0.5 I0.8
G1 X2.95
G17
G3 X3.75 Y0 J0.8
X3.69 Y-0.65 Z-1.16 I-3.75
X3.52 Y-1.28 Z-1.82 I-3.69 J0.65
X3.25 Y-1.87 Z-2.47 I-3.52 J1.28
X2.87 Y-2.41 Z-3.13 I-3.25 J1.87
X2.41 Y-2.87 Z-3.79 I-2.87 J2.41
X1.87 Y-3.25 Z-4.45 I-2.41 J2.87
X1.28 Y-3.52 Z-5.11 I-1.87 J3.25
X0.65 Y-3.69 Z-5.76 I-1.28 J3.52
X0 Y-3.75 Z-6.42 I-0.65 J3.69
X-0.65 Y-3.69 Z-7.08 J3.75
X-1.28 Y-3.52 Z-7.74 I0.65 J3.69
X-1.87 Y-3.25 Z-8.4 I1.28 J3.52
X-2.41 Y-2.87 Z-9.05 I1.87 J3.25
X-2.87 Y-2.41 Z-9.71 I2.41 J2.87
X-3.25 Y-1.87 Z-10.37 I2.87 J2.41
X-3.52 Y-1.28 Z-11.03 I3.25 J1.87
X-3.69 Y-0.65 Z-11.69 I3.52 J1.28
X-3.75 Y0 Z-12.34 I3.69 J0.65
X-3.69 Y0.65 Z-13 I3.75
X-3.52 Y1.28 Z-13.66 I3.69 J-0.65
X-3.25 Y1.87 Z-14.32 I3.52 J-1.28
X-2.87 Y2.41 Z-14.97 I3.25 J-1.87
X-2.41 Y2.87 Z-15.63 I2.87 J-2.41
X-1.87 Y3.25 Z-16.29 I2.41 J-2.87
X-2.34 Y-2.93 Z-16.5 I1.87 J-3.25
X2.34 Y2.93 I2.34 J2.93
X-2.34 Y-2.93 I-2.34 J-2.93
X-1.22 Y-2.8 I0.5 J0.62
G1 X-0.97 Y-2.49
X-0.87 Y-2.37 Z-16.48
X-0.78 Y-2.25 Z-16.44
X-0.69 Y-2.14 Z-16.37
X-0.62 Y-2.05 Z-16.27
X-0.55 Y-1.97 Z-16.14
X-0.51 Y-1.91 Z-16.01
X-0.48 Y-1.88 Z-15.86
X-0.47 Y-1.87 Z-15.7
G0 Z15

in autoreport.h i changed the following to get realtime positioning information:

[image]
[Configuration_files.zip](https://github.com/MarlinFirmware/Marlin/files/10290564/Configuration_files.zip)

### Bug Timeline

It started when i switched to the faster mainboard.

### Expected behavior

I expected it to move in a helical toolpath every time as it should.

### Actual behavior

Sometimes it crashes because it suddenly moves somewhere else.

### Version of Marlin Firmware

Marlin 2.1.2

### Printer model

Milling machine

### Electronics

MKS TinyBee

### Bed Leveling

None

### Your Slicer

None

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

GeneralSnipes, 2023-01-01:
I think I am also running into the same issue but on a 3D printer. The last thing I turned on was arc support and since then my print will randomly stop and sit there with the hot end still on and sitting inside the part unable to receive any commands until it is restarted. I found that it only happens during arc movements or arc segments of test prints and I was able to see one error pop up on my screen saying it was unable to read a line of gcode and showed that the line had no corresponding value following the G. Only G (X Pos) (Y Pos). I am running on a BTT SKR E3 Mini V3. Figured I'd add this comment here to say it appears to also happen randomly on 3d prints.

shortedcircuit, 2023-01-18:
I did some more testing and discovered one little problem with the (self programmed) gcode sender software. But solving this issue (it was sending too many lines and somehow seemed to corrupt the input buffer) did, unfortunately, NOT solve the G2/G3 issue. I have 2 MKS TinyBees (one in the Machine, one in the software lab for developing) and both have the issue.

Additionally I discovered an other issue with the TinyBee and/ or Marlin: 
Machine was at position X31 Y-31 and should do a diagonal move G1 X21 Y-21 but instead the outcome is a crooked movement to something like X29.xxx Y-27.yyy but the TinyBee shows the position as X21 Y-21.

I bought a BTT SKR 3, loaded the same software and all of the problems are gone. 

So apparently Marlin on a TinyBee isn't working for me.

closed 2023-03-31 by github-actions[bot]