MarlinFirmware/Marlin #25140: [BUG] Nozzle diving-scraping into bed at printjob

Fixx, 2022-12-23:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

It started with Marlin 2.1.2, 2.1.1 and older bugfix versions I don't have this issue. When I tried to manually move Z to below 0, it won't go, and Z0 is really aligned with bed so my bed leveling parameters are on spot(same parameters as older fws).

But when I start a print via SD, printer;
1-Do G28 flawless,
2-Do G29 J flawless
3-Then it dives below Z0 and scrapes bed very badly.

I also tried without G29 J or added M420 S1 but same issue again.

How can I troubleshoot this issue?

### Bug Timeline

Start with 2.1.2

### Expected behavior

Nozzle won't go below Z0.

### Actual behavior

Nozzle go below Z0 and scrapes bed.

### Steps to Reproduce

Start a print via SD card.

### Version of Marlin Firmware

2.1.2

### Printer model

Rapide Lite 200

### Electronics

Stock RL200 RUMBA board

### Add-ons

Inductive Proximity Sensor

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin 2.1.2.zip](https://github.com/MarlinFirmware/Marlin/files/10292684/Marlin.2.1.2.zip)

Fixx, 2022-12-26:
I troubleshoot this issue a bit more; and issue is G0/G1 codes for Z axis executing wrong. I can home X/Y/Z and move X/Y/Z via LCD without any issues. I can execute G28 and G1/G0 X/Y via pronterface without issues, but when I send G0 Z10, it makes rapid movement, strange sounds and also two Z steppers got out of sync and printer stops, not executing any more G0 Z command but can execute G0 X/Y commands.

My other issue #25139 most probably related with this Z stepper problem.

I revert back to my 1 year old configured Marlin 2.0-bugfix and printer works flawlessly.

ellensp, 2022-12-26:
Each move has a feed rate associated with it

First please note feed rates are capped to the max available for that axis regardless of F value used.
eg  #define DEFAULT_MAX_FEEDRATE          { 200, 200, 30, 80 }
 
If the gcode line for a move does not have a F parameter it uses the last seen F parameter

when using the lcd it, thelcds send a built in F parameter, 
when using pronterface it uses the feedrates on the screen as the F parameter

See the XY feedrate and the Z feedrate 
[image]

So when you click +X10 , pronterface sends G0 X10.0 F3000
And when your click +Z10, pronterface sends G0 Z10 F100

You need to lower the DEFAULT_MAX_FEEDRATE Z value (30 in the above example) to a value what you machine can actually do. So that requests from the LCD or Pronterface cannot exceed the machines capabilities.

Fixx, 2022-12-26:
Changing DEFAULT_MAX_FEEDRATE { 200, 200, 30, 80 }  to { 120, 120, 20, 50 } solves this problem and issue #25139 . Also changed XY_PROBE_FEEDRATE (60*60) from default 133*60 value.

But at previous firmwares(2.1.1 and 2.0.x-bugfix) those higher values work without an issue? I still think there is some acceleration or feedrate change on this version.

closed 2023-03-10 by github-actions[bot]