MarlinFirmware/Marlin #25142: [BUG] (bug summary) MKS Robin Nano 3.0 weird Y movement

Ram-85, 2022-12-23:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hi,

I compiled the new 2.1.2 firmware for my Sapphire Plus Corexy running on MKS Robin Nano V3.0, firmware compiled successfully, but when testing movement all move fine except for Y- , the print head moves with Y+ but not with Y-, the printer homes for X and Z but not for Y. I am not sure what's going on, I recompiled everything but still the same results!

Attaching the config files here
[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/10296295/Config.Files.zip)

### Version of Marlin Firmware

2.1.2

### Printer model

Sapphire Plus

### Electronics

MKS Robin Nano v3.0

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-12-23:
You have #define ENDSTOPS_ALWAYS_ON_DEFAULT
Which means that it always checks endstosp even with moves, vs the default behavior of only  checking them while homing,

I would guess if you do a M119 (Endstop status), you will find that Y min Endstop is triggered 

Manually trigger this endstop and send a new M119, does it now say open? in which case you just need to change
Y_MIN_ENDSTOP_INVERTING to true

Interestingly I notice your config has changed
#define X_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING true // Set to true to invert the logic of the endstop.

but not Y_MIN_ENDSTOP_INVERTING

Ram-85, 2022-12-24:
Thank you for the great catch and advice, this totally crossed me! I think the issue is solved although I took the whole printer apart to mod the print head. Will let you know if it is still there after reassembly!

closed 2022-12-24 by Ram-85