MarlinFirmware/Marlin #25145: [BUG] Extruder motor just stops working after a few layers

LaurenceGough, 2022-12-24:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hello,

I recently upgraded to a new Extruder and also Marlin 2.1.2. I know I shouldn't make two changes at once but this issue I do not believe is due to a fault of the Extruder. Before these two changes I have been printing fine for months.

I have calibrated Extruder e steps. Not a hotend flow issue. 

There isn't always the nasty sound as in the attached small clip. Afterwards, the extruder is dead until the printer is powered off and back on. Same issue every print over a few layers.

The K factor test print file always prints just fine.

I have tried with and without Linear advance - same issue.

I ran some test g code through it with insane extruder speeds both in retraction and normal and the extruder was fine.
I am using a 1.4vref for the Extruder which was how it come out of the box and has always served me just fine with the standard extruder. Motor never gets hot.

Has anyone else seen this?

Many thanks

### Bug Timeline

Today

### Expected behavior

Print as usual

### Actual behavior

Extruder stopped a few layers into a print every time, sometimes with an odd noise sometimes not.

### Steps to Reproduce

Print a print job with more than one layer and the problem happens every time.

### Version of Marlin Firmware

2.1.2

### Printer model

Creality Ender 5 Pro

### Electronics

All stock - 4.2.2 board.

### Add-ons

OMG Extruder v2s f1 + Creality Spider 300C Pro

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/10299237/Marlin.zip)

https://user-images.githubusercontent.com/82294976/209452779-4a693a7c-f232-45e1-b3a2-b380137ab07e.mp4

[ringing_tower.zip](https://github.com/MarlinFirmware/Marlin/files/10299238/ringing_tower.zip)

Bob-the-Kuhn, 2022-12-25:
Sounds like something is either loose or is getting too hot.

If I'm not mistaken the stepper drivers are soldered onto the controller.  You could go into the pins file and swap the E and X pins , swap the cables and see if the problem stays with the stepper driver or the stepper motor.

LaurenceGough, 2022-12-25:
> Sounds like something is either loose or is getting too hot.
> 
> If I'm not mistaken the stepper drivers are soldered onto the controller. You could go into the pins file and swap the E and X pins , swap the cables and see if the problem stays with the stepper driver or the stepper motor.

Thanks.

I've had a wobble of all cables and even tried tapping the Extruder motor parts etc with no hesitation at all whilst running an extended extruder test movement.

It stops working, requiring a full printer power cycle no matter the print. It is printing along with some of the best quality prints I've seen to date, perfection with no skipping, gaps or anything but then the extruder just suddenly goes. It seems more often than not it doesn't make any noise like it did in that video above.

I've had a bit of success today, I think I'm going to be able to finish a small test print right now!

I reduced all extruder accel and jerk max settings right down to stupidly low values just to see what would happen.

Perhaps these geared extruders need quite a low setting on these values. The stepper motor is cold always.

I suspect it's hitting some kind of driver safety stop - due to how a printer power cycle is required? 

If this print or others fail with these slow settings I'll try swapping the pins, good suggestion thanks.

I am struggling to find recommended values with geared extruders (3:1). I may have to just do my own trial and error.

Many thanks and Merry Christmas.

LaurenceGough, 2022-12-26:
Turning down accel seems to have resolved the issue so I will close this. Looks like the 3:1 geared extruder combined with my stepper motor driver doesn't like too high accel values causing it to go into a protection mode.

With a value of 500 for the extruder motor accel it printed a small test print fine which was failing before.

closed 2022-12-26 by LaurenceGough