MarlinFirmware/Marlin #25148: Weird Syntax error in modules not regularly edited.

inspectionsbybob, 2022-12-25:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

When Compiling 2.1.2 just downloaded, I am getting syntax errors in files That are directly downloaded and are not usually editable.  I checked and had a successful compile of 2.1.1 with edits for my configuration.  I did a file compare on Config.h and Config_adv.h and copied all of the relevant changes.

Here is the terminal output for the two errors
.
.
.
Compiling .pio\build\mks_robin_pro\src\src\feature\runout.cpp.o
In file included from C:\Users\Bob\.platformio\packages\framework-arduinoststm32\cores\arduino/WString.h:29,
                 from C:\Users\Bob\.platformio\packages\framework-arduinoststm32\cores\arduino/Print.h:27,
                 from C:\Users\Bob\.platformio\packages\framework-arduinoststm32\cores\arduino/Stream.h:26,
                 from C:\Users\Bob\.platformio\packages\framework-arduinoststm32\cores\arduino/HardwareSerial.h:29,
                 from C:\Users\Bob\.platformio\packages\framework-arduinoststm32\cores\arduino/WSerial.h:5,
                 from C:\Users\Bob\.platformio\packages\framework-arduinoststm32\cores\arduino/wiring.h:47,
                 from C:\Users\Bob\.platformio\packages\framework-arduinoststm32\cores\arduino/Arduino.h:36,
                 from Marlin\src\feature\../sd/../inc/../HAL/./STM32/../shared/Marduino.h:36,
                 from Marlin\src\feature\../sd/../inc/../HAL/./STM32/HAL.h:28,
                 from Marlin\src\feature\../sd/../inc/../HAL/HAL.h:30,
                 from Marlin\src\feature\../sd/../inc/MarlinConfig.h:31,
                 from Marlin\src\feature\../sd/cardreader.h:24,
                 from Marlin\src\feature\runout.h:28,
                 from Marlin\src\feature\runout.cpp:31:
Marlin\src\feature\runout.cpp: In function 'void event_filament_runout(uint8_t)':
Marlin\src\feature\../inc/../../Configuration.h:1840:34: warning: too many arguments for format [-Wformat-extra-args]
 1840 |   #define FILAMENT_RUNOUT_SCRIPT "M600"
C:\Users\Bob\.platformio\packages\framework-arduinoststm32\cores\arduino/avr/pgmspace.h:71:40: note: in definition of macro 'sprintf_P'
   71 | #define sprintf_P(s, ...) sprintf((s), __VA_ARGS__)
      |                                        ^~~~~~~~~~~
Marlin\src\feature\runout.cpp:132:25: note: in expansion of macro 'PSTR'
  132 |       sprintf_P(script, PSTR(FILAMENT_RUNOUT_SCRIPT), tool);
      |                         ^~~~
Marlin\src\feature\runout.cpp:132:30: note: in expansion of macro 'FILAMENT_RUNOUT_SCRIPT'
  132 |       sprintf_P(script, PSTR(FILAMENT_RUNOUT_SCRIPT), tool);
      |                              ^~~~~~~~~~~~~~~~~~~~~~

followed by this error a few minutes later

Compiling .pio\build\mks_robin_pro\src\src\module\temperature.cpp.o
Marlin\src\module\settings.cpp: In static member function 'static void MarlinSettings::reset()':
Marlin\src\module\settings.cpp:3145:7: error: expected '}' before ';' token
 3145 |       ;
      |       ^
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:39,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\settings.h:28,
                 from Marlin\src\module\settings.cpp:46:
Marlin\src\module\../inc/../../Configuration.h:684:29: note: to match this '{'
  684 |     #define DEFAULT_Kd_LIST { 55.97, 61.55
      |                             ^
Marlin\src\module\settings.cpp:3141:11: note: in expansion of macro 'DEFAULT_Kd_LIST'
 3141 |           DEFAULT_Kd_LIST
      |           ^~~~~~~~~~~~~~~
In file included from Marlin\src\module\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\module\settings.h:28,
                 from Marlin\src\module\settings.cpp:46:
Marlin\src\module\../inc/../core/macros.h: In instantiation of 'constexpr decltype ((lhs + rhs)) _MAX(L, R) [with L = signed char; R = unsigned int; decltype ((lhs + rhs)) = unsigned int]':
Marlin\src\module\settings.cpp:3274:67:   required from here
Marlin\src\module\../inc/../core/macros.h:436:20: warning: comparison of integer expressions of different signedness: 'const signed char' and 'const unsigned int' [-Wsign-compare]
  436 |         return lhs > rhs ? lhs : rhs;
      |                ~~~~^~~~~
*** [.pio\build\mks_robin_pro\src\src\module\settings.cpp.o] Error 1

Most of the other error were caught by the Sanity tester that would say things like:

 if you have **feature X** 
you need to enable **flag1** or **Flag2** 

and then I could search for Flag1 or Flag2 and find what I forgot to enable.  No such luck on these two.
 

Thoughts/ideas appreciated ...

PS... I could not find a page with a zip file for 2.1.x-BugFix but since this is less than a week old....

### Bug Timeline

Just downloaded 2.1.2

### Expected behavior

I expected a clean compile or warnings pointing to Config.h or Config_adv.h.

### Steps to Reproduce

Load in 2.1.2
Load in 2.1.1
Compare my Config.h and Config_adv.h and moved edits over
Build !

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/10299283/Configuration_adv.zip)

### Version of Marlin Firmware

2.1.2

### Printer model

M=JGMaker Artist-D

### Electronics

stock electronics, BLtouch and H2 Extruders

### Add-ons

Never got that far

### Bed Leveling

None

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-12-25:
The error is caused by your config file
`    #define DEFAULT_Kd_LIST { 55.97, 61.55 
`
Needs a closing }

ellensp, 2022-12-25:
The warning is also due to a bad config
you have
`#define FILAMENT_RUNOUT_SCRIPT "M600"
`
but if you read the note above
` // With multiple runout sensors use the %c placeholder for the current tool in commands (e.g., "M600 T%c")
`

Ie you NEED
`#define FILAMENT_RUNOUT_SCRIPT "M600 T%c"
`

ellensp, 2022-12-25:
Only one warning remains.

```
In file included from Marlin/src/module/../inc/MarlinConfigPre.h:37,
                 from Marlin/src/module/../inc/MarlinConfig.h:28,
                 from Marlin/src/module/settings.h:28,
                 from Marlin/src/module/settings.cpp:46:
Marlin/src/module/../inc/../core/macros.h: In instantiation of 'constexpr decltype ((lhs + rhs)) _MAX(L, R) [with L = signed char; R = unsigned int; decltype ((lhs + rhs)) = unsigned int]':
Marlin/src/module/settings.cpp:3274:67:   required from here
Marlin/src/module/../inc/../core/macros.h:436:20: warning: comparison of integer expressions of different signedness: 'const signed char' and 'const unsigned int' [-Wsign-compare]
  436 |         return lhs > rhs ? lhs : rhs;

```
This is easily fixed

```DIFF
diff --git a/Marlin/src/module/settings.cpp b/Marlin/src/module/settings.cpp
index aa0cfe52f7..f23bd95e1f 100644
--- a/Marlin/src/module/settings.cpp
+++ b/Marlin/src/module/settings.cpp
@@ -3271,7 +3271,7 @@ void MarlinSettings::reset() {
     #if ENABLED(DISTINCT_E_FACTORS)
       constexpr float linAdvanceK[] = ADVANCE_K;
       EXTRUDER_LOOP() {
-        const float a = linAdvanceK[_MAX(e, COUNT(linAdvanceK) - 1)];
+        const float a = linAdvanceK[_MAX((uint8_t)e, COUNT(linAdvanceK) - 1)];
         planner.extruder_advance_K[e] = a;
         TERN_(ADVANCE_K_EXTRA, other_extruder_advance_K[e] = a);
       }

```

closed 2022-12-25 by ellensp