MarlinFirmware/Marlin #25156: E0 not heating up (Heater will not engage and make the temperature rise)

bryanibarramercado, 2022-12-27:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

When updating firmware to Marling 2.1.2 the board no longer heats up E0 and causes a heating failed error. I switched back to marling bugfix 2.1.x and the problem no longer appears. Unfortunately I was wanting to text out input shaping in marlin 2.1.2. I was wondering if there is any reason this happens? I have tested the same thermistor, and heating cartridge on a previous version and it heats to the appropriate temperature. So it is not hardware related. 

[Configuration_adv.zip](https://github.com/MarlinFirmware/Marlin/files/10307133/Configuration_adv.zip)

### Bug Timeline

Marling 2.1.2

### Expected behavior

Heat up to desired temperature.

### Actual behavior

Did not heat up E0.

### Steps to Reproduce

1. Loaded up firmware via SD card update.
2. Heated 3D printer to 275.
3. Printer does not heat up.

### Version of Marlin Firmware

2.1.2

### Printer model

Custom tronxy X5SA

### Electronics

Octopus pro V1 F429, TMC2209 and TMC2208 steppers, 24V system.

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

ellensp, 2022-12-27:
You have 
  #define CONTROLLER_FAN_PIN 2           // Set a custom pin for the controller fan

This is in conflict with the hotend PA2

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

ellensp, 2022-12-27:
In your own repository for an older marlin you have
  #define CONTROLLER_FAN_PIN FAN2_PIN        // Set a custom pin for the controller fan

where
#define FAN2_PIN                            PD12  // Fan2

closed 2022-12-27 by ellensp