MarlinFirmware/Marlin #25159: [BUG] Sensorless Homing and Skipping steps

JGLSoft, 2022-12-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Sensorless homing only works with stealthChop max speed: 100. Any other setting homing axes XY slams harshly steppers vibrates and few seconds later endstop is triggered.
Stepper motors skip steps with stealthChop max speed: 100 setting. Only works 60 and below. Any current / current_home, home bump, feedrate settings, same result. 
I also came across the setting homing_threshold, can it be included in the configuration files in the later versions that it can be configured?

### Bug Timeline

since ugrade of tmc2209 drivers

### Expected behavior

Homing Axes
Moving steppers without skipping.

### Actual behavior

Either: 
1. Homing Axes normally and skipping steps
Or 
2. Homing Axes harshly few seconds later triggering endstops and stepper motors working normally.

### Steps to Reproduce

1. Homing axis
2. Moving Steppers

### Version of Marlin Firmware

Marlin 2.1.1

### Printer model

MKS SGen_L v1.0

### Electronics

TMC2209

### Add-ons

TL Smoother

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

thisiskeithb, 2022-12-28:
> Add-ons
> TL Smoother

Remove these as they cause the step signals from TMCs to not work correctly.

ellensp, 2022-12-28:
TL Smoother should only be used with DRV8825 stepper drivers.

closed 2022-12-28 by ellensp

JGLSoft, 2022-12-28:
Removed the TL Smoother same result

ellensp, 2022-12-29:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

JGLSoft, 2022-12-29:
If it cannot be configured in the configuration.h  and configuration_adv.h is it not considered as a bug?

thisiskeithb, 2022-12-29:
> If it cannot be configured in the configuration.h and configuration_adv.h is it not considered as a bug?

Use `*_STALL_SENSITIVITY` / [`M914`](https://marlinfw.org/docs/gcode/M914.html) to tune those values.

I have a couple sensorless homing builds here with TMC2209s and 5160s and they can home fine with the correct configuration.

This isn’t a support forum, so please use the links posted by ellensp for configuration troubleshooting.

JGLSoft, 2022-12-30:
_STALL_SENSITIVITY does not work it's just like on or off switch there is no in-between. It either slams hard or does not move at all. I already tried different current values including home_current still the same. The only thing that's working is the one I mentioned above with any current values (either not both). That's why I came here coz it can't be configured thru the config files. I came across the homing_threshold, is this tied with the hybrid_threshold? If it can be configured I think this might work.

JGLSoft, 2022-12-30:
[https://github.com/MarlinFirmware/Marlin/issues/24764](url)