MarlinFirmware/Marlin #25160: [BUG] Linear advance with i2s (MKS Tinybee and TMC2209)

nikybiasion, 2022-12-28:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I've tried the bugfix 2.1.x but when i print the linear advance calibration test, only the first line with K=0 is printed, in the other lines the motor extruder not spin.
With this pull request https://github.com/HoverClub/Marlin/commit/b5acd17db79a1ffa360f595486b707dcec021762, first line was printed fine, in the second line the extruder works, but the print of the line does not stop, X motor still spin to the bed endstop, i need to turn off the printer.
I use TMC2209_STANDALONE e MKS Tinybee

### Version of Marlin Firmware

Busgfix 2.1.X

### Printer model

MKS Tinybee

### Bed Leveling

ABL Linear grid

### Your Slicer

Other (explain below)

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[config.zip](https://github.com/MarlinFirmware/Marlin/files/10316179/config.zip)

javlonsodikov, 2023-02-14:
Hello @thinkyhead Any update on this issue?

boldenko, 2023-04-13:
same shit here. any updates?

nikybiasion, 2023-05-20:
Still not working in 2.1.2.1

pedr0cks, 2023-07-03:
Mine worked fine. 
If you download the files above you'll have to uncomment the Linear advanced config in the advanced configuration file.
line 2108

nikybiasion, 2023-07-08:
what is your hardware configuration?

javlonsodikov, 2023-09-07:
Please keep this open

javlonsodikov, 2023-12-07:
Is this issue fixed?

nikybiasion, 2023-12-08:
Not fixed yet :(

javlonsodikov, 2024-03-08:
Please keep it opem

binleo, 2024-05-09:
Have any update for this case?

PanosPetrou, 2024-08-16:
Still exists...

closed 2024-11-29 by github-actions[bot]