MarlinFirmware/Marlin #25170: [BUG] Random min temp errors at bootup with software SPI max31865

truglodite, 2022-12-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am running a GTR v1.0 board with an external max31865 for e0, configured with software SPI. Since upgrading to 2.1.2 the printer randomly halts right after bootup, showing low e0 temp errors. It seems 50/50 when I reset if it will succeed booting up, or give another low temp error. Once it is booted up properly, it prints as expected with no issues. This did not happen on 2.1.1 or earlier.

Unplugging/replugging vs using the reset button does not appear to make a difference in the chance of this happening. I thought maybe something changed with how soft SPI is initialized; where sometimes the max is not "ready" early enough at bootup and the error gets triggered?

### Bug Timeline

2.1.2

### Expected behavior

Boot up every time without issues.

### Actual behavior

Low max31865 temp error halt right after bootup more than half the time.

### Steps to Reproduce

Plug in the printer, or hit the reset button.

### Version of Marlin Firmware

2.1.2

### Printer model

a8

### Electronics

gtr v1.0, 2209's, no m1 expansion

### Add-ons

max31865 spi breakout with 3-wire pt100

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

I included my modified the pins file along with my config files files:
 
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/10326713/configs.zip)

ellensp, 2022-12-31:
enable 
#define MAX31865_IGNORE_INITIAL_FAULTY_READS 10 // Ignore some read faults (keeping the temperature reading) to work around a possible issue (#23439).

truglodite, 2022-12-31:
Thank you for the advice. I tried your suggestion and it appears to have fixed the problem. I was able to do 5 power off resets and 5 reset button resets without an error.

closed 2022-12-31 by truglodite