MarlinFirmware/Marlin #25171: Behavior Question - gets lost after several aborted prints

inspectionsbybob, 2022-12-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I have noticed that after many aborted prints (stop print) that the printer starts to get "lost" and its behavior becomes less predictable.

The resulting behavior is varied from wandering axis movement during extruder/bed warm up to "pauses" during prints for no reason.  (Invoking a "change filament" and immediately continuing will normally either finish the print or at least continue for a while...

Anyway, is there a M-code to do a "reset" or "cleanup" that will return all the stacks and queues to a clean state/slate?

### Bug Timeline

NA

### Expected behavior

NA

### Actual behavior

Inconsistent after many failed (stopped) prints

### Steps to Reproduce

Start print
Stop print
Start Pint
Stop Print
Start Print
.
.
.

### Version of Marlin Firmware

current bug-fix

### Printer model

JGMaker Artist-D

### Electronics

NA

### Add-ons

NA

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

Mjankor, 2022-12-31:
Had something similar happen last night when I aborted a print during the first couple of moves. It appeared to randomly reset the home position. 

Couple of quick questions. Are you using input shaper?
What Gcode does your printer run on abort print? G28XY? G27?
Running from SD Card?

inspectionsbybob, 2023-01-02:
No to the input shaper...
I am running repetier-server.
If I do a "stop" it does just that...STOP with no movement after it executes or stops sending codes.
I have to manually send it a G1 Z100 to get the hot-end away from the aborted print so I don't think there are any gcodes/script in the "stop" section.  The stop on Repetier doesn't do a clean retract like a pause, I think it just stops sending gcodes which makes sense as it is "safer" until you figure out what is going on.

cbagwell, 2023-01-02:
When you see its moving, does the firmware also update the screen or does M114 or similar report the movement as well or does it stay fixed location?

inspectionsbybob, 2023-01-02:
There is no X/Y motion in the Start-code.  this is my "Start-Gcode"

G91                                                                   ; relative moves until we know where we are
G1 Z5 F5000                                                     ; lift nozzle
M104 S200                                                      ; start heating Nozzle
M117 Waiting On Bed Temp
M190 S[first_layer_bed_temperature]            ; set bed temp AND WAITfor BED temp to be reached
     **((((( here is about where the Hotend starts "wandering )))))**
G28 ; home all axes after Bed to Temp
M420 S1                                                         ; turn bed leveling back on after G28
.
.
.  much more about printing the purge line and so follows, but everything after the G28 is fine (so far)

The G1 Z5 is probably why I have never seen any Z-movement.

closed 2023-03-14 by github-actions[bot]