MarlinFirmware/Marlin #25174: [BUG] Latest bugfix doesn't recognize SD card, and print freezes

rq3, 2023-01-01:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Compiled and installed latest (Jan. 1, 2023) bugfix using the latest example configuration files for the Anycubic Predator. 

The changes to configuration.h are:
1) Change from Classic_ui to Color_ui
2) Turned on endstop_interrupts (as I normally do anyway), to suppress new compile warning
3) Turned off MESH_EDIT_GFX_OVERLAY, as this has always thrown an error

Selecting a file to print results in a question mark (?) in the center of the screen, and the touch functions are frozen. Removing and re-inserting the SD card then causes Marlin to re-boot, the file is selectable, and the print starts.

However, after about one minute of printing, the print and touch screen freezes.

### Bug Timeline

New, as of bugfix Dec.31, 2022, or maybe earlier?

### Expected behavior

I expect the SD card to be read correctly, the print to start, and complete.

### Actual behavior

The SD card requires that it be removed and re-inserted, which re-boots Marlin. The print is then able to be started, but the printer completely freezes after roughly a minute. A power cycle is required to recover.

### Steps to Reproduce

1) Compile and install the latest bugfix, using the latest exampe/delta/anycubic/predator configurations
2) Try to print from the SD card

### Version of Marlin Firmware

bugfix 2.1.2 as of January 1, 2023

### Printer model

Anycubic Predator

### Electronics

stock Trigorilla_pro

### Add-ons

none

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/10329608/Configuration.zip)

rq3, 2023-01-02:
Just tried the very latest bugfix (Jan. 2, 2023). SD card is recognized and starts print UNLESS the card is accessed via the touchscreen icon, in which case the central question mark and printer freeze still happens.

Once a print is started, it still freezes after about 1 minute. Requires a power cycle to force re-boot.

jmz52, 2023-01-02:
What platformio environment was used for firmware compilation?
What MCU is installed on your Trigorilla pro board?
Have you tried to print with TFT_CLASSIC_UI?

rq3, 2023-01-02:
> What platformio environment was used for firmware compilation? What MCU is installed on your Trigorilla pro board? Have you tried to print with TFT_CLASSIC_UI?

1) Non-Maple environment
2) STM32F103ZE MCU
3) Haven't tried classic in a LONG time. I'll give it a shot tomorrow and report back

rq3, 2023-01-03:
Observations:
Classic_UI (non-Maple) works fine, with no issues at all.

1) Color_UI goes into a repetitive boot loop at the boot screen if the SD card is in place at first start.
2) Attaching the SD card from the touch icon results in a question mark in the center of the display.
3) Attaching the SD card from the menus works if the SD card is removed, inserted, and then attached.
4) During a print the printer either freezes after about a minute, OR randomly wanders off in the -Y direction after about a minute.

One other note. A compiled firmware.bin under earlier bugixes (Nov.16, 2022) was 422kb. Under the lates bugfixes it is 299kb.

jmz52, 2023-01-03:
Are there any error message on COM port, like "sd read error" (with Color_UI) ?

rq3, 2023-01-03:
> Are there any error message on COM port, like "sd read error" (with Color_UI) ?

Nope. Monitoring over USB with Repetier Host seems normal. It even sees the contents of the SD card, until a print is actually started. When the freeze happens, there is no communication at all. At least Repetier shows nothing. Everything just stops dead, and requires a power cycle to recover.

jmz52, 2023-01-04:
Please to print from SD with SOFTWARE_SPI instead of hardware SDIO and Color_UI:
- disable SDIO_SUPPORT
- enable SOFTWARE_SPI
- change `#define SDSS PD2` to `#define SDSS PC11` in pins_TRIGORILLA_PRO.h file. You may have "Media Init Failed" error at startup, in this case remove and insert SD card so it can be initialized correctly. Alternatively you can insert SD card after Marlin is booted.

If print is successful, please provide photo of MCU on your Trigorilla pro board and board's revision information.

uncellon, 2023-01-04:
Hmm, on my Flying Bear Ghost 5 (Robin Nano v1.2, XY drivers is TMC2208, ZE - A4988) I have a similar behavior - Y randomly moves on printing a random distance if Marlin built with TFT_LVGL_UI. With TFT_COLOR_UI all works fine. I monitored the COM port using Pronterface and there are no any errors. Y offsets appear between 1 and 25 minutes according to my observations
[image]

jmz52, 2023-01-04:
> This results in a repetitive white screen boot loop.

@rq3, does it boot if SD card removed?
Or still white screen boot loop even with no SD card?

rq3, 2023-01-05:
In an attempt to track down when this all started, I compied with the exact same configuration files:
1) Marlin Release 2.1.2. Treats the SD card correctly, but freezes after a short time while printing.
2) Marlin Patched 2.1.2. Screws up the SD card as above, and freezes after a short time while printing.
3) The firmware.bin I am currently using was apparently compiled from a very late bugfix for 2.1.1, dated 12/16/2022, and it works fine in all respects.

rq3, 2023-01-12:
Bumping this. The SD card issue still exists with today's bugfix (January 11, 2023). Latest bugfix, latest example configurations for delta/anycubic/predator, with Color-UI.

This is a bug.

rq3, 2023-01-16:
Bumping this. The SD card issue still exists with today's bugfix (January 15, 2023). Latest bugfix, latest example configurations for delta/anycubic/predator, with Color-UI.

This is a bug.

Roxy-3D, 2023-01-20:
I agree there are some issues with the SD cards at the moment.   I have been seeing problems with the RepRap Full Graphics LCD panels and the usage of SD cards on them.   I am starting to think it isn't the SD Code.  I'm starting to suspect the SPI interface code.

Does anybody have any thoughts or comments about my thinking?

rq3, 2023-01-22:
More info. I tried 3 different SD cards. One acted dead, as if it were a piece of paper. The second re-boots the printer when removed, and then re-inserted, with no ability to print from it. The third can be read IF the card is refreshed several (like 6) times, and a print can be started, which then locks up after about a minute. A power cycle is required to recover.

Reverting to bugfix2.1.2 prior to Dec. 31, 2022 all three cards, and the printer, work fine.

thisiskeithb, 2023-01-22:
Would you be able to test some [`bugfix-2.1.x` commits](https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.1.x) to narrow this down?

The strategy is to find a commit from some point in the "distant" past where the feature works. Then, test a commit from halfway between that date and today…. And then you keep going to the commit half-way in between your "known to work" commit and your "known to be broken" commit until you find the exact day where it broke.

If you started from a point 256 commits in the past, it would take no more than 8 tests to find the exact commit that broke it.

rq3, 2023-01-22:
@thisiskeithb, I'd be happy to do a binary search IF you can tell me clearly and explicitly HOW to download the zip file for each commit.

Best Regards

thisiskeithb, 2023-01-22:
It's actually a command if you have git installed: https://git-scm.com/docs/git-bisect

Start with a minimal config, otherwise conflicts will be painful to resolve.

---

To do it manually, you'd browse to each commit from the main [`bugfix-2.1.x` commit history](https://github.com/MarlinFirmware/Marlin/commits/bugfix-2.1.x) using the <kbd>**< >**</kbd> button on the right:

[image]

Then you'd click on the green <kbd>**< > Code ▼**</kbd> button and "Download ZIP":

[image]

jmz52, 2023-01-22:
It looks like there are two problems intertwine here.

First is the easy one - Anycubic printer stuck when tries to print from SD card (with SDIO_SUPPORT) with TFT_COLOR_UI and works file with TFT_CLASSIC_UI. This mimics the behavior of Kingroon KP3 boards with GD32F303 MCU, so I expect the root cause to be the same - flawed DMA controller than can't handle two concurrent data transfers. This is a common problem for clones and fake STM32. There is a workaround (already tested by Kingroon users) and I'll include it to STM32H7 code I am working on (ETA - a week or two).

Second problem is "SD init fail" issue. Kingroon users reported this problem during workaround development and as far as I know it is not related to TFT code. It is easier to reproduce this problem when SOFTWARE_SPI is used instead of SDIO_SUPPORT, but I did not dig into this problem.

Please note that MSG_MEDIA_INIT_FAIL message is displayed by `LCD_ALERTMESSAGE()` and is not replaced by MSG_MEDIA_INSERTED / MSG_MEDIA_REMOVED messages which are displayed by `LCD_MESSAGE()`.
I.e. if you re-insert SD card after getting MSG_MEDIA_INIT_FAIL message, the message will stay on screen even after CD card is correctly inited and mounted.

rq3, 2023-01-22:
The bug starts in commit aaae56A, titled "TFT_COLOR_UI asyn DMA SPI" addressing issue #24980 on December 12, 2022.
I am currently printing successfully with the commit just prior to that.

A binary search was done from November 15, 2022 to December 16, 2022. In all cases, the attached configuration files were used, and three different SD cards were used to verify that the card themselves were not at fault.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/10475138/config.zip)

rq3, 2023-01-24:
> It looks like there are two problems intertwine here.
>
> First is the easy one - Anycubic printer stuck when tries to print from SD card (with SDIO_SUPPORT) with TFT_COLOR_UI and works file with [...]

The Anycubic Predator I am using for development is about 5-6 years old. Old enough so that it is highly unlikely that the STM32F103ZE chip is a fake or clone, but you never know.

I do know that I have done some surface mount modifications to the motherboard over the years (reducing the Fosc resistor value to "zero" ohms to eliminate stepper glitches, and removing the ADC input filter components as they interfered with a PT1000 amplifier, and were not correct anyway). 

There is absolutely no way these mods, which were done years ago, affect SD or SPI functionality, and during that time I took great effort to verify the origin and pin-out of the MCU. If it is a fake or clone, it would have had to have been manufactured very, very quickly after the release of the ST Micro original, and externally and functionally it is indistinguishable from the real thing. Even under the microscope I use for surface mount soldering. At the time, Chitu, the manufacturer of the motherboard, was attempting to break into the printer market, and appeared to take great pains to keep their quality high, even if it cost them money in the short term. Indeed, they did cut costs later in the game, but I am not aware of them ever resorting to clones or fakes.

Also, be aware that there is no commonality between a Chitu (Anycubic) Trigorilla board, and a Chitu (Anycubic) Trigorilla Pro board.

closed 2023-01-24 by rq3