MarlinFirmware/Marlin #25179: Duplication Mode using more than 2 heads.

SlickNickeL, 2023-01-02:
### Is your feature request related to a problem? Please describe.

This has been an issue for some time now and has been addressed under #11702 01/09/2018 and merged pull request #13373 11/02/2019.  Unfortunately, several turn of events have made it appear that a solution was found and implemented, along with supporting code and an “OK”to test the final results. I checked the operation on our machines, and it would never engage 3 print heads at the same time, only 2.

### Describe the feature you want

There is a need for duplicating 3D printers in manufacturing. The STACKER S4 is one example using Repetier firmware. Marlin has as yet, to accomplish this task.  

The duplicating machine consists of 3 or more heads attached to a single X carriage.
All gcode is cloned from E0 to E1, E2, E3, E4 etc. Temperature should be able to be set independently per head. All extruder feeds are cloned from E0.  X & Y movement commands are all from E0.  

Applications are well established in the manufacturing and hobby industry. Large runs of small pieces and parts as well as usable first run products, greatly benefit from this duplication process. Model railroad, jewelry, pins, funnels, rods, caps and many other applications are possible. At www.pitchouts.com we uses one such printer to produce three separate colored units, in their entirety at the push of one button. 

### Additional context

FUTURE ADVANCEMENTS: The firmware that can offset each individual head extrusion rate, to adjust for a individual extrusion multiplier, will have solved and completed the golden equation in the manufacturing realm.

ellensp, 2023-01-03:
"I checked the operation on our machines, and it would never engage 3 print heads at the same time, only 2."

Then you must have some sort of example Config file. which would be useful to provide, so we can see what does or does not get defined in code...

ellensp, 2023-01-03:
Looking at what I can see about this printer online

This printer, its not a IDEX, (only a single X axis) it allows you to manually set the offsets of each hotend and then it prints from multiple extruders.

Can you confirm this is the case?

I presume it allows for each hotend to have a different temperature? 

What about E steps? are they independent? or simply mirrored across all extruders?

SlickNickeL, 2023-01-03:
Your presumptions are correct.All heads are fixed (or movable) on the X axis. 1-6 heads all printing at the same time. There really doesn’t need to be an offset registered in the software because it sees only one hotend E0 which then clones the E steps to the other filament steppers.Note: careful attention is needed in order to keep the heads from crashing into completed work on the build plate. This is completely a users responsibility. Several slicers can compensate for this. The E steps are all cloned (mirrored) from E0. Each hot end would have the ability to have a separate temperature. This is how Repetier is set up.Be something special if E steps were all independent, but from my last conversations, it would take some serious coding and an 8bit machine couldn’t do it.  (4 years ago)Sent from my iPad

ellensp, 2023-01-04:
Issue seems to be a simple gcode passing issue

M605 S2 P7
echo:Duplication mode: ON ( 0 1 2 )
This works as expected, all 3 extruders, 0,1 and 2 move as one

But 
M605 S2 E2   (which should be the same)
echo:Duplication mode: ON ( 1 2 )

Results in only extruders 1 and 2 being on, 0 is disabled  (as per the list)

SlickNickeL, 2023-01-04:
If I remember correctly, I could get all 3 extruders to purge at the same time just to test and that worked. When placed the code in the file for printing, it would only engage 2 extruders.  Was kind of weird that a single extruder advancement would work, but wouldn’t work on a print job. One extruder would always be off.

ellensp, 2023-01-04:
Please test this

In Marlin/src/gcode/control/M605.cpp

replace
`      else if (parser.seenval('E')) duplication_e_mask = pow(2, parser.value_int() + 1) - 1; // Set the mask by E index`
with
`      else if (parser.seenval('E')) duplication_e_mask = _BV(parser.value_int() + 1) - 1; // Set the mask by E index`

This works on my test environment.

M605 S2 E2  
> echo:Duplication mode: ON ( 0 1 2 )

G1 E20 and extruders 0,1  and 2 extrude as expected

M605 S2 E1  
> echo:Duplication mode: ON ( 0 1 )
G1 E20 all extruders 0 and 1 extrude as expected

SlickNickeL, 2023-01-04:
Question: I am setting up a machine (Rumba) that has 3 dedicated E steppers one for each hotend.  
Will Marlin compensate the E-Steps set in EEPROM per E stepper  or will it clone the EEPROM from E1 settings and use only this settings across all E steppers?
Example: I pulled a Repetier firmware unit from service and I’m in the process of transferring the eeprom settings for Marlin.
REPETIER EEPROM
Extr.1 87.750 steps per mm
Extr.2 87.750 steps per mm
Extr.3 80.000 steps per mm

I never noticed this mistake? My non conclusive notes say we did an experiment on this and I’m guessing it was never returned to pre test conditions, but works as planned. Did Repetier clone the EEPROM mm per step from Extr.1 to 2&3?  I assume it does since this machine has done many different colors and runs, and all the parts meet specs.

Will Marlin do the same?

My hypothesis is Repetier clones all information from EEPROM and E steps only on the first extruder, and passes all information blindly to any activated extruder. 
will Marlin do the same?

Roxy-3D, 2023-01-12:
Do we have any updates on what has been figured out?    I am very interested in knowing if the current code does things correctly or if we need to make some small tweaks.

ellensp, 2023-01-12:
@Roxy-3D as far as I can tell its work fine.    But im just using a controller and 3 stepper as 3 E's  Not a real machine

SlickNickeL, 2023-01-13:
**Nice Job Ellensp-3D and thank you Roxy-3D**
This works!
We have our first MARLIN 3 head manufacturing 3D printer. So much simpler than Repetier.  
We will run our first full production shortly at www.pitchouts.com
Lots more we can do with this and still some questions, and future coding to be done, to complete the golden formula. 
I will make my Config.h and .adv available either here or in the next step under a different feature request, since this was really a bug fix. This can be closed out.

closed 2023-01-13 by ellensp