MarlinFirmware/Marlin #25184: 64128 display vertical menu arrangement? [FR] (feature summary)

pedroscsant, 2023-01-04:
### Are you looking for hardware support?

MKS Robin E3D with MKS Mini v3 Display

### Describe the feature you want

Would be nice to be able to mount any kind of 64x128 graphical displays on the vertical, if possible. 
The trouble I'm having is that some parts of the text are cut beyond the size of the display. 
Is there any way to resize the text or rearrange the menu order so it will be displayed properly? 

Please, see pics attached.

[image]
[image]

Nuck-TH, 2023-01-04:
While status screen may be usable if modified for this layout, menu system will be borderline unusable - for every menu item only one word would fit and you would need to cut item names short or scroll every of them, which, imho, both cases would be very frustrating and slow to use.

pedroscsant, 2023-01-05:
> While status screen may be usable if modified for this layout, menu system will be borderline unusable - for every menu item only one word would fit and you would need to cut item names short or scroll every of them, which, imho, both cases would be very frustrating and slow to use.

Yes, I guess you are right, I just read the documentation properly and realised it would be pretty much unreadable.

Using the 64p as the horizontal arrangement, with the current 6x9 caracthers, you would be able to fit 10 caracthers max.
Scaling it down to something like a 4x6 you would fit 16. That would look pretty weird already and still far from the 21 we have on the 128p with the 6x9 arrrangement.

So unless you come up with smaller words, abreviations or even symbols to replace menu entries like "advanced configuration", i guess it's not worth the trouble.

Nonetheless, I still think a "slim" vertical menu would be pretty cool.

closed 2024-06-02 by thisiskeithb