MarlinFirmware/Marlin #25187: [BUG] (Only 1x Z axis motor homes - Dual Y + Dual Z SKR 1.4 Turbo)

pman92, 2023-01-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

I am compiling marlin for a dual-Y and dual-Z axis CNC router build (V1 Engineering Lowrider V3) with 2 separate endstops on  each. Z axis homes towards the max direction. 
X and Y axis work properly as expected (with multi Y endstop), including homing. 

Due to Z homing towards the max direction, I have changed Z_HOME_DIR to +1 (from -1)
I have defined "Z_MULTI_ENDSTOPS". 
When I set "Z2_USE_ENDSTOP" to "_ZMAX_" (with underscores), I receive a compile error:
`Marlin\src\module/endstops.h:93:39: error: 'Z_MAX' was not declared in this scope`
I would of thought this should work, as the relevant pins file includes:

> #elif ENABLED(Z_MULTI_ENDSTOPS)
  #ifndef Z_MIN_PIN
    #define Z_MIN_PIN                      P1_27  // Z-STOP
  #endif
  #ifndef Z_MAX_PIN
    #define Z_MAX_PIN                      P1_00  // PWRDET
  #endif 

If I set the "Z2_USE_ENDSTOP" to "_XMAX_" (which I am not using, as there is only 1x X axis stepper), Marlin compiles OK and the Z axis moves OK when I use "move axis" from the menu. Both steppers move in the correct direction. 
However when I try and home Z (G28 X), only the first Z1 stepper moves. The second Z2 stepper (connected to E1 driver) does not move, but it is energized (I cannot spin it by hand). The gantry crashes if I let it home too long and get too far out of square.

I have checked all the endstops with M119. I am shown:

> Reporting endstop status
x_min: open
y_min: open
y2_min: open
z_min: open
z_max: open
z_probe: open 

Since Z homes towards the Z-max direction, I would expect a z_max and z2_max (similar to the Y axis), instead of z_min. 

All endstops work and report TRIGGERED when I hold my finger on them. 

**IMPORTANT:** The Z2 endstop (shown as z_max from M119, which I set as "_XMAX_") is still connected to the Z_MAX_PIN as defined in the pins file (P1_00 pin, PWRDET, next to ZMIN connector). If I connect it to the X_MAX_PIN (P1_26) it does not work. 

 

### Bug Timeline

It was also present in Marlin 2.1.1 release

### Expected behavior

I expected it to compile with the Z2 endstop set as ZMAX (due to its definition in the pins file), M119 to display a "z1_max" and "z2_max" endstop, and for both Z steppers to move when homing.

### Actual behavior

I had to change the Z2 endstop to XMAX for it to compile, and only 1 stepper moved when homing

### Steps to Reproduce

I have attached config files to reproduce 

### Version of Marlin Firmware

Latest Marlin bugfix 2.1.x, downloaded from github 1 hour ago

### Printer model

V1 Engineering Lowrider V3 CNC Router

### Electronics

BTT SKR 1.4 Turbo / TMC2209 Stepper drivers

### Add-ons

Manual bed probe is enabled

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config Files Marlin 2.1.x bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/10342011/Config.Files.Marlin.2.1.x.bugfix.zip)

ellensp, 2023-01-04:
You have #define Z_HOME_DIR 1, which uses z_max 
you cannot set Z2_USE_ENDSTOP to  `_ZMAX_`  as z1 is already using it!

Set 
`  #define Z2_USE_ENDSTOP   _ZMIN_
`

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- 3D Printing Discord server. Join link: https://discord.gg/3dprinters
- Marlin specific Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2023-01-04 by ellensp

pman92, 2023-01-04:
Thank you. I didn't realise changing the homing direction also changed the endstop that is used, but it makes sense. 
Maybe a comment in the configuration files near the X/Y/Z_HOME_DIR mentioning it could help.