MarlinFirmware/Marlin #25189: [BUG] LCD doesn't work on Melzi targets

mikebignell, 2023-01-04:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The LCD doesn't work on Melzi board targets because it's missing the define `IS_MELZI` from `pins_MELZI.h`

### Bug Timeline

New issue, introduced with commit 9949672f1d303d786e20cd053701fcd5c61de736

### Expected behavior

LCD should work

### Actual behavior

LCD does not work

### Steps to Reproduce

1. LCD does not work

### Version of Marlin Firmware

FIRMWARE_NAME:Marlin bugfix-2.1.x (Jan  4 2023 13:20:55)

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

mikebignell, 2023-01-04:
I'll submit a PR with a fix shortly

ellensp, 2023-01-05:
Closing as a PR has been added

closed 2023-01-05 by ellensp