MarlinFirmware/Marlin #25192: G29 J Not Working Correctly

500cubes, 2023-01-05:
### Did you test the latest `bugfix-2.1.x` code?

No, but I will test it now!

### Bug Description

G29 J tilts the mesh too much creating an usable mesh. 

Original 3 points are. Bottom Middle, Top Right Corner, Top Left Corner. I tried to modify the 3 point pattern from a triangle to a straight line and it still tilted the mesh towards one side. I also tried changing the direction of the 3 point triangle with no success. I up loaded a previous build, v2.0.9.3 and G29 J worked great. 

At this point i have no idea what else it could be. 

### Bug Timeline

After installing 2.1.2

### Expected behavior

G29 J to probe the 3 points and slightly tilt the mesh

### Actual behavior

Mesh before
[image]

Mesh After G29 J. You can see it is tilted more towards one side and the mesh values are different
[image]

### Steps to Reproduce

1. G29 L0
2. G29 J
3. View Mesh results in LCD. 

### Version of Marlin Firmware

2.1.2

### Printer model

Custom CORE XY

### Electronics

SKR2.0 TMC2209

### Add-ons

BL TOUCH

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

### Additional information & file uploads

[Config Files.zip](https://github.com/MarlinFirmware/Marlin/files/10348856/Config.Files.zip)

titanrw, 2023-01-14:
Maybe related to:

https://github.com/MarlinFirmware/Marlin/issues/24665

2.0.9.3 was also the last one that worked for me with G29 J

klack, 2023-01-19:
@500cubes Pardon, but what screen are you running?

500cubes, 2023-01-22:
> 

Sure looks like it!

500cubes, 2023-01-22:
> @500cubes Pardon, but what screen are you running?

its a BTT TFT70

thisiskeithb, 2023-03-03:
Fixed in https://github.com/MarlinFirmware/Marlin/pull/25453

closed 2023-03-03 by thisiskeithb