MarlinFirmware/Marlin #25193: [BUG] (bug summary) M25

Caitex212, 2023-01-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Hey,
I got a problem, the following code should move the printhead on Z up a bit, pause and then continue printing.
Instead it's moving up, moving down and then pauses.

; STEP 1 : Move the head up a bit when Z-Hop is enabled
G91;Switch to relative positioning
G0 Z1;Move Z axis up a bit
G90;Switch back to absolute positioning
; STEP 2 : Pause the print
M117 Print paused...
M25 ;Pausing print...
; STEP 3 : Repositioning on the Z axis
G91;Switch to relative positioning
G0 Z-1;Move Z axis back down
G90;Switch back to absolute positioning
M117 Printing...

### Expected behavior

I expected it to move up, pause, move back down.

### Actual behavior

It moved up, back down and then paused.

### Version of Marlin Firmware

Unkown (Printer doesn't react to M115)

### Printer model

Alfawise U30

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

GMagician, 2023-01-05:
Check #24877, it may solve

Caitex212, 2023-01-06:
Ok, but the problem is not, that my printer isn't pausing or doesn't unpause. The problem is that it executes lines after the M25 before the actual pause...

GMagician, 2023-01-06:
> The problem is that it executes lines after the M25 before the actual pause...

I also noticed this. In my case printer move to pause and immediately return to print and stop after a while (depends on queued elements I suppose). Once pause no way to resume

closed 2023-03-19 by github-actions[bot]